exelban/stats #4: Feature request: the disk monitor should show a usage chart similar to the one for RAM and CPU

sirburpalot, 2019-07-01:


exelban, 2019-07-01:
@sirburpalot thanks for propose.
Added in [v1.2.1](https://github.com/exelban/stats/releases/tag/v1.2.1).

sirburpalot, 2019-07-02:
@exelban This isn't what I was proposing. What I meant was that the chart should show the rate at which data is currently being read from/written to the disk, not the amount of free space.

exelban, 2019-07-02:
@sirburpalot i wanted to do it from the beginning.
But for this i must implement different readers for the module. 
Also, menu for this will be a little bit difficult. 
So for the beginning i need to do a new menu view. After that i will move to few readers in one module.

I will leave this issue opened till do this.

exelban, 2020-08-09:
Add the option to show Disk IO stats in the [last](https://github.com/exelban/stats/releases/tag/v2.2.1) version.

closed 2020-08-09 by exelban