exelban/stats #6: Feature request: display the names of the indicators in the Chart/Chart with value modes

sirburpalot, 2019-07-01:
The names could be displayed vertically in order to save space (see the attachment).
[image]

sirburpalot, 2019-07-01:
Or, better yet, place the caption closer to the chart, like this.
[image]

exelban, 2019-07-02:
@sirburpalot hmm. Thanks. I will try to implement this in a few days.

exelban, 2019-07-04:
@sirburpalot 
[image]
[image]

It will be in next release.

exelban, 2019-07-05:
[v1.2.2](https://github.com/exelban/stats/releases/tag/v1.2.2)

closed 2019-07-05 by exelban

sirburpalot, 2019-07-05:
@exelban One more suggestion: you should have probably used a monospace font like Menlo for this, as the "E" in "MEM" clearly looks out of place. It would also be a good idea to make the font weight identical for the mini and chart modes - right now, the chart modes use a bold font, whereas the mini mode uses a normal font. Either one of them looks fine on its own, but the different modes should use the same font weight for consistency.