exelban/stats #20: New logo

exelban, 2019-09-04:
The existing logo was just found somewhere in the internet.
I will be very helpful if someone will help with the new logo.

MUI68K, 2020-05-30:
You may use this icon if you want - I made it this evening when I spotted the request:
[image]

You may of course also decide not to use the icon. I needed something to do and so I opened up Sketch.
[Stats.zip](https://github.com/exelban/stats/files/4706432/Stats.zip)

ahydronous, 2020-06-28:
I actually think the current icon is pretty decent. It doesn't really show anyway, since the app is not in the dock. Or do you think you will get rightsholder issues?

exelban, 2020-06-29:
Yeah, the existing icon is ok. But only because I don't have an option. It does not look great.

I just like the nice icons. It's even good to have a nice one to use it in App Store (when app will be deployed).

Also maybe it will help me with app design (help needed too) because I jumping between different styles and cannot stop on one.

jaykovski, 2020-10-11:
Proposed icon:
[image]

exelban, 2020-10-11:
@jaykovski hi. Yeap, it looks nice.

jaykovski, 2020-10-12:
@exelban I have made a pull request #110

exelban, 2020-10-12:
@jaykovski thanks for the new icon.

closed 2020-10-12 by exelban