exelban/stats #26: Average Energy Impact at 5%

sohocoke, 2019-09-14:
Over the last few days I’ve been observing Stats ranking 2nd the energy impact list.

#21 would probably help with this, but maybe it’s also worth looking at other ways to improve CPU usage. E.g. can data for the Overview and Top Processes sections only be acquired when the popover is shown?

exelban, 2019-09-15:
@sohocoke, as I see in Activity Monitor Stats consume ~2-3 points. And almost all of them is 2 top processes (which are using to read CPU and Memory usage) and netstat process (for network) usage. Do you have the same numbers?

sohocoke, 2019-09-15:
What do mean by points?

I only have CPU and disk plugins enabled, and I see 1 entry for ‘top’ when I expand the entry for Stats under Energy Impact. There are no detailed numbers in the sub entry for ‘top’ however.

exelban, 2019-09-15:
It looks like the activity monitor shows the battery consumption in percentage (I'm not sure). Yeah, if you have only CPU and disk plugin you must see only one top process. And what power consumption you have (of Stats app)?

sohocoke, 2019-09-15:
As I said, the Average Energy Impact has been around 5%. I’ve seen it to be as bad as 13%.

exelban, 2019-09-15:
Hmm, it's very interesting. I have averaged around 1%. And it's with all modules enabled.

sohocoke, 2019-09-15:
What it looked like for the whole day:

[image]

exelban, 2019-09-16:
Wow, it's not normal. I already added the option to choose an update interval for the readers. It can help in some way.
But still, it's strange. And what top process usage you have (inside Stats)?

exelban, 2019-09-17:
ok. I rewrite readers for Top Processes and now only for Memory module I using `top`. For CPU usage I switched to `ps`. Also now this processes runs only when the view is opened. I believe it will improve power consumption.

[v1.3.4](https://github.com/exelban/stats/releases/tag/v1.3.4)

closed 2019-09-17 by exelban

sohocoke, 2019-09-17:
Averaging at < 1% for the last few hours since updating. This is an excellent update, thanks a lot. 👍

exelban, 2019-09-18:
@sohocoke thanks for reporting.