exelban/stats #28: Menu bar item is blank when starting Stats with menubar hidden

ukyoi, 2020-04-11:
Thanks for reading this issue report. Recently I use Stats everyday and I found if I launch Stats with "automatically hide and show menu bar" turned on, the item of menu bar of Stats will be totally blank. I'm not familiar with macOS developing and can't figure out why just by walking through the source code. Could you help me to check this issue?

macOS Version: 10.15.4
Step to Reproduce:
1. Quit Stats Completely.
2. Turn on "Automatically hide and show the menu bar" in Preference.
3. Open Stats again.

exelban, 2020-04-11:
@ukyoi thanks. Fixed in [v1.6.1](https://github.com/exelban/stats/releases/tag/v1.6.1)

closed 2020-04-11 by exelban

ukyoi, 2020-04-12:
It works like a charm. Thanks a lot.