exelban/stats #30: Crash after app launch

brutella, 2020-04-29:
The app closes automatically after some seconds and reports that dispatch queue `com.repeat.queue` crashed.

[Crash Report](https://github.com/exelban/stats/files/4550436/Stats_2020-04-29-100409_Matthias-MacBook-Pro-16.crash.zip)

exelban, 2020-04-29:
@brutella thanks for the report. I will take a look today.

exelban, 2020-04-29:
There is not so much information about the crash in the report. But it looks like its a problem with a repeater. I have changed my own implementation to the 3d-party library. And I hope it will fix the crash. Give me to know please if it helps.

New version: [v1.6.3](https://github.com/exelban/stats/releases/tag/v1.6.3)

brutella, 2020-04-29:
Unfortunately this doesn't resolve the crash.

Here is a new crashlog →  [Stats_2020-04-29-120319_Matthias-MacBook-Pro-16.crash.zip](https://github.com/exelban/stats/files/4551012/Stats_2020-04-29-120319_Matthias-MacBook-Pro-16.crash.zip)

exelban, 2020-04-29:
Thanks for your feedback. 

I never work with .dSYM. But it looks like its generating automatically. I looking now how I can receive more information from your crash.

exelban, 2020-04-29:
@brutella it appears when you just install the app?

Or after changing some app settings?

brutella, 2020-04-29:
The app launches successfully and shows up in the menu bar.
After 2-3 seconds the app just quits.

exelban, 2020-04-29:
Yeap. Thanks. 

I figured out where was the problem. Looks like the disk module cannot read the size of some disk. Can I send you a dev version of the application, to check if it works?

brutella, 2020-04-29:
Sure
matthias.hochgatterer@gmail.com

exelban, 2020-04-29:
Thanks. 
I send you a new version. Please give me to know if something is changed.

exelban, 2020-04-29:
There is a new version.

[Stats.dmg.zip](https://github.com/exelban/stats/files/4551572/Stats.dmg.zip)

brutella, 2020-04-29:
This version works now. 👍

closed 2020-04-29 by brutella

exelban, 2020-04-29:
Thanks for your contribution👍