exelban/stats #31: Catalina 15.4 illegal hardware instruction

rememberlenny, 2020-05-28:
Hi!

Trying to get this to work, and Im seeing the following error in the terminal when running this on Catalina 15.4:

`[1]    13626 illegal hardware instruction  /Applications/Stats.app/Contents/MacOS/Stats`

exelban, 2020-05-28:
@rememberlenny Hi. 
Can You share the full output? Or its only one crash message?

rememberlenny, 2020-05-28:
The actual app crashes silently. The only way to get the app error is running the direct .app binary.

That content is the full error.

—
Leonard Bogdonoff
RememberLenny.com | 949-322-8496

exelban, 2020-05-28:
@rememberlenny thanks. 

What mac do you have?

rememberlenny, 2020-05-28:
*Hardware Overview:*

  Model Name: MacBook Pro

  Model Identifier: MacBookPro16,1

  Processor Name: 6-Core Intel Core i7

  Processor Speed: 2.6 GHz

  Number of Processors: 1

  Total Number of Cores: 6

  L2 Cache (per Core): 256 KB

  L3 Cache: 12 MB

  Hyper-Threading Technology: Enabled

  Memory: 16 GB

exelban, 2020-05-29:
@rememberlenny Hi.

I tried to understand what causes this crash and error. The only reason is missing some instruction maybe. Unfortunately, I cannot reproduce this error. 

If you would like to help, you can run the project in xcode and give information about the crush (I hope xcode will tell what exactly happened).

rememberlenny, 2020-05-29:
That sounds good.

sohocoke, 2020-06-03:
I'm also having crashes since yesterday on 1.6.4. The crash logs are appearing in Console.app; is there a .dsym I can use to symbolicate it?

exelban, 2020-06-03:
@sohocoke [Stats.dsym](https://serhiy.s3.eu-central-1.amazonaws.com/Github_repo/stats/Stats.app.dSYM.zip)

sohocoke, 2020-06-03:
Couldn't symbolicate it (UUID mismatch).
Here are the crash log contents:

```
Process:               Stats [94443]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            Stats
Version:               1.6.4 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [94443]
User ID:               1036

Date/Time:             2020-06-03 16:21:08.604 +0900
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        D0C06868-6645-B57D-432E-E6F6FDE9C13F

Sleep/Wake UUID:       30AC9FB1-E575-4B46-9072-2BB1A0C9AE00

Time Awake Since Boot: 76000 seconds
Time Since Wake:       1100 seconds

System Integrity Protection: enabled

Crashed Thread:        4

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [94443]

Thread 0:
0   ???                           	0x00007fff39a0658d 0 + 140734160201101
1   ???                           	0x00007fff398e3556 0 + 140734159009110
2   ???                           	0x00007fff398e32ae 0 + 140734159008430
3   ???                           	0x00007fff398e31e3 0 + 140734159008227
4   ???                           	0x00007fff3bf85058 0 + 140734199517272
5   ???                           	0x000000010771b35f 0 + 4419859295
6   ???                           	0x0000000107724d8c 0 + 4419898764
7   ???                           	0x0000000107728088 0 + 4419911816
8   ???                           	0x000000010772ec22 0 + 4419939362
9   ???                           	0x000000010772ed5b 0 + 4419939675
10  ???                           	0x00007fff3993035f 0 + 140734159323999
11  ???                           	0x00007fff399302f3 0 + 140734159323891
12  ???                           	0x00007fff39930268 0 + 140734159323752
13  ???                           	0x00007fff3992febe 0 + 140734159322814
14  ???                           	0x00007fff398ff7e2 0 + 140734159124450
15  ???                           	0x00007fff398fec82 0 + 140734159121538
16  ???                           	0x00007fff3bf84a22 0 + 140734199515682
17  ???                           	0x00007fff36b07fef 0 + 140734110924783
18  ???                           	0x00007fff36b07d32 0 + 140734110924082
19  ???                           	0x00007fff36b04df3 0 + 140734110911987
20  ???                           	0x00007fff36b04a39 0 + 140734110911033
21  ???                           	0x00007fff3bfb0246 0 + 140734199693894
22  ???                           	0x00007fff3bfb00b0 0 + 140734199693488
23  ???                           	0x00007fff3acb4092 0 + 140734179786898
24  ???                           	0x00007fff3acb37b9 0 + 140734179784633
25  ???                           	0x00007fff3acaba27 0 + 140734179752487
26  com.apple.HIToolbox           	0x00007fff384a02b8 AEProcessAppleEvent + 54
27  ???                           	0x00007fff36afee4c 0 + 140734110887500
28  ???                           	0x00007fff36afd380 0 + 140734110880640
29  ???                           	0x00007fff36aef09e 0 + 140734110822558
30  ???                           	0x00007fff36ac1465 0 + 140734110635109
31  ???                           	0x0000000107703249 0 + 4419760713
32  ???                           	0x00007fff70fde7fd 0 + 140735089076221
[... 353 lines cut ...]
Foundation                           4K        1 
IOKit                             15.5M        2 
Kernel Alloc Once                    8K        1 
MALLOC                           184.7M       33 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
Memory Tag 242                      12K        1 
STACK GUARD                       56.0M        7 
Stack                             11.0M        7 
VM_ALLOCATE                         68K        8 
__DATA                            23.3M      275 
__DATA_CONST                       245K       12 
__FONT_DATA                          4K        1 
__LINKEDIT                       356.7M        7 
__OBJC_RO                         32.0M        1 
__OBJC_RW                         1780K        2 
__TEXT                           165.8M      270 
__UNICODE                          564K        1 
mapped file                       47.5M       14 
shared memory                      652K       17 
===========                     =======  ======= 
TOTAL                              1.3G      674 
TOTAL, minus reserved VM space   896.2M      674 

```

exelban, 2020-06-03:
@sohocoke thanks. I will debug it today.

exelban, 2020-06-03:
@sohocoke the .dSYM what I send before has the same uuid that v1.6.4
`D666ECBC-AD50-3ABD-812B-E45998D67919`

The crash report is unsymbolicated. And I cannot fetch any information from it(

exelban, 2020-06-03:
@rememberlenny Could you check if you have a crash report in the Console app when running Stats?

sohocoke, 2020-06-04:
@exelban  The crash report shows the UUID of the app's executable as `D666ECBC-AD50-3ABD-812B-E45998D67919`, which matches the output of `dwarfdump --uuid` of the .dsym you posted. Despite this, the crash report doesn't symbolicate -- AFAIK symbolication should happen automatically once the .dsym is indexed by Spotlight, which doesn't happen. MacSymbolicator also reports 'uuid mismatch'.

Any ideas?

exelban, 2020-06-04:
@sohocoke I'm new with crash reports. 
But it looks like the problem with the report. I can open the report in xcode, but it cannot be symbolicated. 

According to [this](https://developer.apple.com/library/archive/technotes/tn2151/_index.html) docs, the report which you share is unsymbolicated. And even more, because in apple example unsymbolicated report has the name of the library which crashed. And in your only `???`.

Could you run the Stats in Xcode? It will give the full stack trace of crash.

sohocoke, 2020-06-04:
It's been a while since I've built from the project... I'll try to make some time soon.

exelban, 2020-06-04:
@sohocoke I can make a little instruction if it will help

sohocoke, 2020-06-04:
Ok, crash is at DiskCapacityReader.swift:85:
```
                        let BSDName: String = String(cString: DADiskGetBSDName(disk)!)
```

DADiskGetBSDName(disk) returned a nil, causing the forced unwrapping to crash. The URL for the disk at time of crash was a mounted network Time Machine volume.

sohocoke, 2020-06-04:
If you can point me to any examples in the codebase of how weird situations are logged or otherwise handled, I can fix and submit a PR.

exelban, 2020-06-05:
@sohocoke https://github.com/exelban/stats/blob/525abaa44315e8989c01ad4454288ba0f4eb9444/Stats/Modules/Disk/DiskCapacityReader.swift#L85.

Thanks for your help. If you want you can just make a check if DADiskGetBSDName exists, and then process the read.

Give me to know if you will do a PR or not. Thanks

exelban, 2020-06-06:
@rememberlenny @sohocoke please, confirm if a new version fix crash.

[v1.6.5](https://github.com/exelban/stats/releases/tag/v1.6.5)

exelban, 2020-06-24:
If the bug still appear, please reopen the issue.

closed 2020-06-24 by exelban