exelban/stats #32: Make the bars different colors

ahydronous, 2020-05-29:
Right now they are all blue, which is slightly confusing if labels are disabled. I like it however, because its a very clean and compact look.  
[image]

If all the bars had slightly separate colors, they would be easier to distinguish.

Another option would be a centered small letter for each type inside the bar. For CPU you would use a C, for RAM a R, for SSD a S, etc.

ahydronous, 2020-05-31:
Now that I thought more about it, another nice option would be to make the bars monochrome (with the possibility for the aforementioned letters inside the bars), so they blend in really well with other macOS menubar icons. I know scope creep is a bad thing, just thought I'd throw it out there :)

exelban, 2020-05-31:
@ahydronous thanks. It's a nice idea. I will try to implement some of them in the next versions.

exelban, 2020-05-31:
@ahydronous there is one problem with putting one letter to bar. 

For example, the bar color is blue and a letter inside is white. When the bar height will be less than a letter, letter will be invisible. Because there will be no background for it. It will be a white letter on the default menu bar background (white in aqua mode).

In theory, I can add this option in the next version, there will be a box with a background. If it will be not a problem, I will add this option.

ahydronous, 2020-06-01:
Hmm, how about this: with the menu bar being in light mode, a dark gray is used (which would contrast with most colors and also the menubar), and when the menu bar is in dark mode, a light grey is used. It’ll take some experimentation, but I think for both the dark and light mode there are sweet spot grays where they contrast both with the bars and the background.

Also, while it is nice for the letters to perfect legible, they will be very small. I think more of them like ‘logos’, as long as the C, R and S are distinguishable from each other it serves the purpose we are after :)

bviefhues, 2020-06-29:
> Now that I thought more about it, another nice option would be to make the bars monochrome (with the possibility for the aforementioned letters inside the bars), so they blend in really well with other macOS menubar icons. I know scope creep is a bad thing, just thought I'd throw it out there :)

+1 from me. I am on dark mode and have my complete menu bar white on black. The blue stats graphs don't fit really well.

exelban, 2020-06-29:
The problem in dark mode with colors is because macOS add some transparency to colors.
I'm going to add the option to specify a color accent for each module individually. It will take some time.

bviefhues, 2020-07-01:
Adding to my earlier comment: Now that we have the RAM memory pressure feature (thanks for this!), would be great if the "green" color could be overridden if that feature is on, i.e. instead of green show the chart with configured color instead of green and switch to yellow or red if memory pressure is high. 

Aim would be to maintain a clean monochrome (black on white or white on black) normal state.

exelban, 2020-07-01:
I'm thinking about the option `Color accent` for each module. Where you can choose colors and the algorithm of colorizing (pressure, depending on load, system color accent, etc). To provide the option where you can specify colors for each level of pressure (for example). I need to create a color system. Where you can create a color palette, and select it where you want. But it's not so obvious for me how it must works.

For example, if someone will want to colorize the CPU usage on 5 different colors (0-20%, 20-40, 40-60, 60-80, 80-100). How it will work, and how the user can specify this range, and map specific colors for each level. It's a little bit confusing me. 

But, for the beginning, I will add the option to select the color for each module from a predefined list of colors. I think it will cover most needs.

ahydronous, 2020-07-01:
I might have an easy solution for the color picking + resource utilization:

- Users only pick a base color
- The bar gets that color
- Resource utilization changes the gradient of the bar the higher it gets
- In dark mode, the bar gets higher RGB values ( = brighter) the higher it gets
- In light mode, the bar gets lower RGB values ( = darker) the higher it gets

The nice thing is that this works for both monochrome and colored bars.

exelban, 2020-07-01:
I don't understand how it works with utilization. Did you propose to change the color brightness depending on resource utilization?

exelban, 2020-07-01:
For me there are different color modes:
- system accent color
- depends on resource utilization (green/orange/red)
- pressure (depend on pressure level)
- some color (specify the color from the list, it the feature it can be color picker)
- monochrome (black/white)

And I would like to implement these options at the beginning.

ahydronous, 2020-07-01:
Those all seem great, then my solution won't be necessary :)

I'm guessing developing the color picker might take some time, so maybe its most useful to first develop the preset colors (system accent, monochrome, pressure level, utilization), and then develop the color picker afterwards? That way, if there are bugs with bar colors they already get spotted ahead of time.

exelban, 2020-07-01:
As I remember Apple provide color picker in SDK, so I don't need to develop my own color picker)

I have one more question, what do you mean monochrome?
Just black on white / white on black or some grayscale colors?
Or there is some monochrome option in macOS?

ahydronous, 2020-07-01:
Yes, black when light mode is active and white when dark mode is active. There are also grey values used for things that are inactive, but those grey values could also be used differently.

Here you can see in light mode
[image]

And dark mode
[image]

If Stats would match with those other icons it would be amazing!

exelban, 2020-07-01:
Yeap, yeap. I can do it. 

I almost finished thinking about how it must work and looks like. Also, I check with pressure if there is no problem changing the colors on the fly. So I think I will develop this option this week. But before I must figure out how I app can be updated to the latest version, without dragging from dmg.

bviefhues, 2020-07-01:
> For me there are different color modes:
>
> * system accent color
> * depends on resource utilization (green/orange/red)
> * pressure (depend on pressure level)
> * some color (specify the color from the [...]

May I suggest, not to make pressure a color mode? For me this is a "warning" or "error" indicator, which should override a color mode (if switched on).

I.e. I configure monochrome color mode, charts are white on black (in dark mode, that is). If memory becomes tight, then the memory chart should flip from white to orange or red. And back to white if memory situation relaxes.

If you implement green/yellow/red as a separate mode, then I am stuck with a green chart on the menu bar.

exelban, 2020-07-01:
@bviefhues wait, it's you asked for pressure levels (#46). And now you don't need it?

exelban, 2020-07-01:
Ohhh, I see. But it will not work in this way. If you choose monochrome, it will be monochrome. There will be no option to change the color to orange or red. Because in this case, it will be not monochrome.

exelban, 2020-07-01:
In theory, it can be called `monochrome until 2 pressure level`. But it will be too complicated.

bviefhues, 2020-07-01:
@exelban I would like to have both :-) 

The runtime color selection logic is probably not that difficult. However, the UI might become quite complex, with all possibilities and options.

exelban, 2020-07-01:
Color selection it's the main problem. Changing the colors in the widget at the runtime it's not a problem at all.

exelban, 2020-07-08:
I add an option to select a color for each module in the [new](https://github.com/exelban/stats/releases/tag/v2.1.7) version.
Also, there is a monochrome option.

closed 2020-07-08 by exelban