exelban/stats #35: Battery status won't show at manubar on v2.0.1

ukyoi, 2020-06-08:
Happy to see Stats comes to v2. While when updated to v2.0.1, the battery status won't show at menubar whether the switch turned on or off. Could you please check that issue?

macOS Version: macOS 10.15.6 Beta (19G36e)
Hardware: MacBook Pro 13'' (2015)

exelban, 2020-06-08:
@ukyoi yeap, checking

exelban, 2020-06-08:
@ukyoi could you check if the widget is showing up when you change the widget to the first one (text only)?

ukyoi, 2020-06-08:
Changing the widget doesn't work, but I found it show again when the battery status changed (I unplug the power adapter).

Thanks for you reply.

exelban, 2020-06-08:
@ukyoi hmm, so the widget shows up when you unplug the power adapter (in that moment macos send a notification that battery property was changed)?

ukyoi, 2020-06-08:
Yes, looks like that. Maybe it is not the app to blame. The power management of my OS often behaviours strangely since it continuously showing my battery need repair.

exelban, 2020-06-08:
ok, please reopen the issue if the problem appears again.

closed 2020-06-08 by exelban