exelban/stats #41: Can't view dropdown while in another app's fullscreen mode

coxjc, 2020-06-23:
I'm not terribly sure if this is fixable or not, but if I'm in fullscreen mode for another app (e.g. Slack), I see nothing when I click on a `stats` icon. It appears to open the dropdown on the desktop that doesn't host a fullscreen app.

exelban, 2020-06-23:
@coxjc Hi. I check this, but I cannot reproduce this problem.
Do you use an external display?

What os version and mac do you have?

coxjc, 2020-06-24:
yeah, i do use two external displays but i also found it occurring when disconnected from both monitors. 

i am running Catalina 10.15.5 on a 15" MBP.

coxjc, 2020-06-24:
actually i found something somewhat odd:

if i am using an external display to the left of my MBP and i click on Stats in the external monitor, the dropdown pops up in the top left of the MBP native display

exelban, 2020-06-24:
yeap, I reproduce this bug. I will fix it in the next release (in a few days).

exelban, 2020-06-24:
Please confirm if it fixed in the new version [v2.1.0](https://github.com/exelban/stats/releases/tag/v2.1.0).

coxjc, 2020-06-24:
@exelban it looks like that release is missing the .dmg

exelban, 2020-06-24:
ohh, forgot. Added.

coxjc, 2020-06-24:
looks good. thank you sir

closed 2020-06-25 by exelban

FallBright6, 2025-10-06:
This issue came up again on my computer. It's a 16" M1 Pro MacBook Pro running macOS 26.0.1

HamlinWang-SH, 2026-10-20:
I can consistently reproduce behavior very similar to this issue with Stats 3.0.22 on macOS 27.0.1, including the external-display symptom described above.

### Steps to reproduce

1. Run Stats with **Combined modules** enabled.
2. Put another application into native macOS full-screen mode. The recording I captured uses the ChatGPT desktop app.
3. Reveal the menu bar and click the combined Stats item to open its details.

### Actual behavior

- macOS automatically switches away from the full-screen Space to a regular desktop on the MacBook's built-in display to show the Stats popup. I do **not** manually switch desktops.
- Clicking Stats on the external display also causes the popup to appear on the built-in display's desktop instead of the display where I clicked.
- This happens every time, rather than intermittently.
- In the recording, the popup briefly appears on the regular desktop and then disappears.

### Expected behavior

The popup should open on the display where its menu bar item was clicked, over the current full-screen application, without switching Spaces or moving focus to another display.

### Environment

- MacBook Air, Apple M4, 16 GB RAM (`Mac16,12`).
- macOS **27.0.1 (26A434)**.
- Stats **3.0.22 (850)**.
- Built-in display is the **main display**, with a logical resolution of 1470 x 956 at 60 Hz.
- External display: **S2716Q**, 2560 x 1440 at 75 Hz, connected using an **HDMI adapter cable**.
- Extended desktop, not mirroring; the external display is arranged above the built-in display.
- **Displays have separate Spaces:** enabled.
- **When switching to an application, switch to a Space with open windows for the application:** enabled.
- **Automatically rearrange Spaces based on most recent use:** enabled.
- **Stage Manager:** disabled.
- Stats **Combined modules:** enabled; Stats **Dock icon:** enabled.
- Magnet **3.0.7** is also running. I have not yet tested with it disabled, so I cannot rule out an interaction.
- Full-screen application in the recording: ChatGPT **26.1013.40934**.

This also resembles #556 and #1267, although I have not established that they share the same root cause. I noticed the later report here that similar behavior returned on macOS 26.0.1.

Would you consider reopening this issue, or would you prefer a new report linked to it? I would be happy to help investigate and test a local fix. Before proposing code changes, I would also like to confirm whether you would welcome a fix for this behavior and how you would prefer it to be submitted given the current PR restrictions.

### Recording

The redacted clip below shows the built-in-display case. The menu bar changes from the full-screen ChatGPT application to the Finder desktop automatically after clicking Stats; I did not manually switch Spaces. The Stats popup briefly appears at approximately 5.2 seconds. Unrelated screen content is masked and audio is removed; playback timing is unchanged. The external-display behavior described above is a separate, consistently reproducible observation.

https://github.com/user-attachments/assets/8deb7d24-8220-4e1a-bb4d-b4e569dd2cf3

jces5307-cmyk, 2026-10-29:
I can reproduce this on Stats 3.0.23 and macOS 27.2 (26B123), single display, Mac18,5 (Apple M6): with a browser in native full screen, clicking a Stats widget slides to the desktop Space and opens the popup there.

I think I found the cause and a one-line fix. `PopupWindow` only sets `.moveToActiveSpace`, so macOS will not place the panel in a full-screen Space and switches Spaces instead. Adding `.fullScreenAuxiliary` lets the popup open on top of the full-screen app:

```diff
 // Kit/module/popup.swift — PopupWindow.init
 self.hidesOnDeactivate = false
-self.collectionBehavior = .moveToActiveSpace
+self.collectionBehavior = [.moveToActiveSpace, .fullScreenAuxiliary]
```

Both module popups and the combined-modules popup use `PopupWindow`, so this covers both. With a local Release build of this change, the popup opens over the full-screen browser and the Space no longer changes. Popups on a normal desktop behave as before. I only tested a single display, so I can't say whether it also fixes the external-display case described above.

The commit is on my fork: https://github.com/jces5307-cmyk/stats/commit/9e1aa2b
PRs are limited to collaborators here, so I'm posting it in this issue. Happy to open a PR if you prefer that.