exelban/stats #43: App crashing right after start

mike4president, 2020-06-28:
When I install the app it open for a second or two and crashes right after. Can't get it to stay open.

exelban, 2020-06-28:
@mike4president could you share a crash report?

kadowa-medien, 2020-06-29:
I have the same issue. Please find my report attached.
Best
Kai
[Stats_2020-06-29-100113_C02TC2BJGTFL.crash.zip](https://github.com/exelban/stats/files/4844537/Stats_2020-06-29-100113_C02TC2BJGTFL.crash.zip)

exelban, 2020-06-29:
@kadowa-medien thanks. Could you try a [new](https://github.com/exelban/stats/releases/tag/v2.1.1) version? And send a crash report if it crashes too?

kadowa-medien, 2020-06-29:
All versions chrash. I installed the latest dmg 2.1.1. I also tried (after
crashing and removing) the homebrew installation. Same result.

I'm on the way and will send you later the reports

Best
Kai

kadowa-medien, 2020-06-29:
here is the screenshot of the brew install and the crash report.
[image]
Report
[Crash.zip](https://github.com/exelban/stats/files/4845069/Crash.zip)

kadowa-medien, 2020-06-29:
And here the report of the installation out of the download behind the 2.1.1 dmg:
[2-1-1_Crash_of_dmg_download.zip](https://github.com/exelban/stats/files/4845093/2-1-1_Crash_of_dmg_download.zip)

Hope that helps
Kai

mike4president, 2020-06-29:
sorry for the late reply. Here is my crash report.

[Stats.txt](https://github.com/exelban/stats/files/4845159/Stats.txt)

Same happens with 2.1.1

weghorst, 2020-06-29:
same for me. please see attached .

[Stats_2020-06-29-111100_Michaels-iMac.txt](https://github.com/exelban/stats/files/4845308/Stats_2020-06-29-111100_Michaels-iMac.txt)

I have also Little Snitch installed

Best regards, Mike

exelban, 2020-06-29:
Thanks, I will debug it today.

exelban, 2020-06-29:
You all have the same crash. And it's related to the network module. 
@mike4president use a little snitch. And it can be related.
@kadowa-medien @weghorst did you have some VPN or other tools for the network?

exelban, 2020-06-29:
ok, it's not related with VPN. It's a problem with accessing the window before it initialized.

exelban, 2020-06-29:
Could someone check if this version is not crashing?

[Stats.dmg.zip](https://github.com/exelban/stats/files/4846271/Stats.dmg.zip)

mike4president, 2020-06-29:
still crashing...:-(

[Stats_2020-06-29-170802_Mikes-iMac.txt](https://github.com/exelban/stats/files/4846357/Stats_2020-06-29-170802_Mikes-iMac.txt)

mike4president, 2020-06-29:
I am also using nextdns.io for dns based blocking. Besides Little Snitch I have Malwarebytes, iStatMenus and Mac Fan Control installed... could they interfere?

mariusv, 2020-06-29:
I also have the same issue and I'm using Little Snitch

mike4president, 2020-06-29:
i removed Little Snitch... Stats ist still crashing

exelban, 2020-06-29:
nope, it's not related to the network interface. It crashes when trying to update a popup window.
Could someone try this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/4846442/Stats.dmg.zip)

kadowa-medien, 2020-06-29:
@exelban , I have cloudflare as DNS provider in my router settings. The provided version also crashs.
Report here:

[Stats_2020-06-29-172549_C02TC2BJGTFL.txt](https://github.com/exelban/stats/files/4846445/Stats_2020-06-29-172549_C02TC2BJGTFL.txt)

exelban, 2020-06-29:
It's not related to your network configuration. I have a DNS AdBlock too)

mariusv, 2020-06-29:
I just tried this [release](https://github.com/exelban/stats/issues/43#issuecomment-651192815) and it works

mike4president, 2020-06-29:
same here... works now

kadowa-medien, 2020-06-29:
Looks good on my site as well! Thank you for this very quick and great support. What was the issue?

exelban, 2020-06-29:
I suppose all of you has a `unknow (WiFi)` under Network in network popup?

mariusv, 2020-06-29:
yeah, that's what I wanted to say. But I'm not using WiFi but a dongle for wired connection and even before was reporting it wrong :-)

exelban, 2020-06-29:
there were 2 bugs: a network reader trying to update a popup window before it was initialized. And the second one related to wifi network name. The first one it a bug. But the second one it's very strange. It looks like mac cannot define a wifi network name.

mike4president, 2020-06-29:
my Wifi is disabled and I am only using a wired connection

exelban, 2020-06-29:
@mike4president and you see `Ethernet` under Network in network popup?

mike4president, 2020-06-29:
> @mike4president and you see `Ethernet` under Network in network popup?

is this what you mean? 
[image]

mariusv, 2020-06-29:
Same here, WiFi disabled but 
[image]

mike4president, 2020-06-29:
this is what I see

[image]

exelban, 2020-06-29:
hmm, it's super strange. I use an external library ([Reachability](https://github.com/ashleymills/Reachability.swift)) to detect connection type and notify when the network state is changed. It looks like they have some problems with determining what connection type is using.

exelban, 2020-06-29:
It looks like it's a good moment to move from this library to my own implementation of these features.

exelban, 2020-06-29:
Ok, for now, I will deploy a new version with this fix. And with time, I will fix the problem with wrong connection type detection.

exelban, 2020-06-29:
Thanks everyone for feedback and help.

closed 2020-06-29 by exelban