exelban/stats #44: App Store support (name problem)

exelban, 2020-06-28:
I would like to deploy the application to the App Store. But there is the app called Stats already.
So there are two options:
- select a new name for the app and rename the app
- I can deploy the app with mStats name to App Store, but use the Stats on github

The first option will be better. But I cannot choose a good name for the app. So I need some help with it.
The second option a little bit confusing. App will have two different names (mStats in App Store, and Stats on github).

ahydronous, 2020-06-28:
I did a quick little bit of brainstorming.

What Stats.app is is basically an app showing system information in the menu bar. That condenses to System Information Menu Bar. Now a little bit of word trickery:

- SysInfoMenBar - nah..
- SysInfoBar - it's ok but not attention grabbing
- SysBar - ah, better
- InfoBar - again ok, I like SysBar more
- SyBar - short and it has a slight tongue-in-cheek wordplay of 'Cyber'

You can also look at it like this, Stats helps you spy / peek on system info, so let's try it:

- SystemSpy - sounds like a full scanner app instead of just a menu bar resource monitor
- SysPeek - nah..
- PeekSys - nah again..
- PeekBar - This sounds pretty good
- SpyBar - This sounds pretty good as well, although 'Spy' sounds a bit more sinister

I laid out my thought process so you have a lot of choice or maybe draw inspiration and come up with another name.

For me I'd personally go for PeekBar / Peekbar. It's short, and the name immediately tells you that helps you take a quick look (in this case system resources) at something via the menu bar.

exelban, 2020-06-29:
Hi, thanks.

`SyBar` sounds good. The only problem that I have it's writing. `sybar` | `SyBar` | `syBar` | `Sybar`. Hmm, maybe it's not soo bad as I think. Maybe there is something with only one word or work with some one-letter prefix.

`PeekBar` a little bit the same problem that previous one.

I'm looking for something which can say that this is the app for some monitoring/statistics for your mac. And at the same time say that this is a menu bar application. And `PeekBar` it's more like just fine app name. And maybe it's even better than my approach.

If we could not found any one word name. There will be just two options: combination of two or just a nice name.

iwittkau, 2020-07-11:
Here are some more words that could be part of a new name

- gauge
- observer
- tracker
- measure

Brainstorming results:

- mGauge or mGage (similar to "engage")
- Gauge Bar
- Stats Observer
- System Tracker
- Measure Menus

I did not check if any of those already exist.

exelban, 2020-07-11:
Hi, thanks for this list.

gauge and mGage its nice, but used already(
The another one is a little bit strange.

exelban, 2020-07-11:
I'm thinking about `statd` name. It's available in the app store. Also, there is a domain name available.
But yes, I know that there is a linux tool with the similar name. And `d` in tools means demon. But still, it's quite a nice name.

qkdxorjs1002, 2020-08-26:
- Pulse
- Cardiogram (ECG)
- MacCardiogram
- WatchMog (Watch'Mac'Dog)

majonez, 2020-08-27:
HardwareStat

0alpha, 2020-09-04:
-1 to `statd`, because of the name confusion. I'm partial to SyBar. `mBar` is SEO hell, as it's a unit of air pressure. Hmm.

yukimuon, 2020-09-07:
`statd` is a linux service, so I don't think it's a good idea.
I really want a name that is distinctive, not like the existing iStat which most people recommend to me and I would recommend Stats back to them.
My opinion, how about `iDashboard` `iDash`?

exelban, 2020-09-08:
Thanks everyone for proposes. 
I know that `statd` is a linux service. But is a short and nice name. And because it's a linux process I reject this name.

As I say before. It can be just a nice word. This `hardware`, `monitor` with different suffix/prefix looks a little bit strange.

Uffje, 2020-09-09:
What about just `Peek` and if your want to add bar I would use `Peekbar` or `peekbar`.

exelban, 2020-09-09:
There is an app called Glance. `Peek` is a nice one, but unfortunately, it's used.

yukimuon, 2020-09-10:
How about `Glimpse`? Or more tech `Ammeter`

exelban, 2020-09-26:
Hmm, Glimpse is a nice one. It's a little bit hard to write if you don't know.

I just figured out that MacMetrics is available.

exelban, 2020-10-20:
Thanks everyone for proposes. 
I will keep the app called `Stats`. And it will be not published to App Store in near future. Even if at some moment I will decide to deploy the app, it will require disabling some modules or some functionalities (because of the sandbox).

closed 2020-10-20 by exelban