exelban/stats #45: Use Sparkle updater framework

ahydronous, 2020-06-28:
The current update method is probably very lightweight, but it's a bit janky by having to restart the app myself and having to unmount and delete the .dmg. Not to mention, there is always a chance for hidden bugs.

With [Sparkle](https://sparkle-project.org/) it will take care of everything for you. It will automatically tell when there is an update, the user can choose if he wants auto-update or not, and the app will update itself and then restart without any leftover .dmg files!

exelban, 2020-06-29:
I look up a few time to Sparkle. But it looks too complex for me (maybe I'm wrong).
For me, it's much easiest to create a new tag on git and that's all. I can add the script which will automatically move the app to the needed folder, and clean up the `.dmg`.

It looks like I must take some time to play with Sparkle, and maybe the process of deploying the new app can be automated.

exelban, 2020-07-01:
I add the script which automatically updates the app. Will be in the next release.

closed 2020-07-05 by exelban

ahydronous, 2020-10-21:
@exelban the issue where Stats creates a Stats.dmg file in my downloads, and does not remove it has started cropping up again.

I still think adding Sparkle is the way to go. Yes, it means a one time big effort, but it also means you won't constantly have to go back to fix these edge case issues, write your own update checker, file remover, etc.
Hell, imagine if there is a unforeseen bug in the file remover and it nukes someone's Downloads folder.

exelban, 2020-10-21:
Hi.

I understand your point. But I will not move to Sparkle. I don't want to have any 3d party dependencies (only if there is no another option). Maybe, I just don't like the way how Sparkle works.

Yoshi2889, 2021-01-24:
I know this issue is closed and you've probably made up your mind, but I do ask that you revisit this. Either way I'm dropping my two cents here.

Especially now that there's been multiple releases in a really short period, the built-in updater has gotten pretty annoying in that it uses persistent notifications and does not show any sort of progress. Sparkle on the other hand provides auto-updates with progress and changelogs (if desired) out of the box using the well-known "An update is available" dialog used in many apps. This creates consistency and familiarity.

Moreover I think the notion of not liking third party dependencies is respectable but also rather dangerous. Sparkle is well-tested and uses secure methods of downloading files when possible. I don't see file verification in your updater code, for instance. You're also relying on a shell script to do the actual update, found inside the unverified DMG file. If ever your github credentials leak and some malicious release drops, your entire userbase could be compromised this way.

Finally the integration would allow apps like https://github.com/mangerlahn/Latest to report updates to Stats and apply those along with other Sparkle apps.