exelban/stats #46: Show RAM memory pressure

bviefhues, 2020-06-29:
macOS activity monitor shows a memory pressure graph with red/yellow/green colors. Would be great for stats to visualize the yellow and red as well in the menu bar. For my Mac, entering the "yellow zone" is a significant performance impact and I would like to know that.

exelban, 2020-06-29:
Sounds great, I will try to combine it with @ #32.

exelban, 2020-06-30:
I never use memory pressure. So I'm not familiar with how it works. But it looks like apple provides a level of memory pressure. And it can be read from `sysctl`. So I just read this value, and colorize the chart is different colors:
- Level 1: green
- Level 2: green
- Level 3: green

You can enable this option for bar and line chart widgets.
This option available in the [new](https://github.com/exelban/stats/releases/tag/v2.1.3) version of app.

closed 2020-06-30 by exelban