exelban/stats #48: wrong temperature readings

mike4president, 2020-06-29:
Hi, I am getting wrong temperature reading with my quad core iMac Late 2017. 

208°C seems a bit hot ;-)

Can Stats report CPU PECI?

and it only reports 2 cores and show the other two when I activate "show hyper threading cores"
[image]

[image]

exelban, 2020-06-29:
Yeap, it looks like your mac has a different CPU temperature sensor code. I don't know if there is a sense of this temperature under this module. It added when Stats does not have a sensors module. And now when you can see the temperature on a specific sensor there is no sense to have this sensor there. 

CPU 1 and CPU 2 it's not a different core. 
[image]

CPU core sensors look like this.

There is no sensors list for each mac. So I just trying to find all possible options on the internet. It looks like I missing sensors for iMac. If you want to help to improve the app and add missing sensors for your mac. I can provide a little instruction how to fetch a list of sensors that your mac has. And next, I will add this sensors to the list.

exelban, 2020-06-29:
There is must more easiest way to collect missing sensors. Because of the app collection all keys in the system, and trying to map it with my list of keys. So I just need to enable the log when key not found on the list. And it will be enough to provide a log from `Console.app`. If you will be interested to help. Give me to know, I will provide a debug version of app to collect the missing keys in logs.

mike4president, 2020-06-29:
Sure thing. Just tell what I have to do.

exelban, 2020-06-29:
Ok, this is a developing version (but it notarized throw apple as normal build).

Before opening the app do next:
- open `Console.app`
- select your mac on device list (by default it must be opened on this tab)
- in the search field enter `Missing sensor`

Now just start the application. In the console, you must see some rows with information about the missing sensor.

You can export these logs to file, and share it. Or just select the logs, copy them and paste there.

[Stats.dmg.zip](https://github.com/exelban/stats/files/4847229/Stats.dmg.zip)

exelban, 2020-07-01:
You can use the latest version, it has enabled missing sensors keys logging.

exelban, 2020-08-02:
Closed due to inactivity. I update a sensor in the popup. But if someone still sees strange temperature, please reopen this issue.

closed 2020-08-02 by exelban