exelban/stats #50: Controll fans (SMC?)

TymurP, 2020-06-29:
It would be a great feature to control fan speed.
Very nice app btw! ;-)

exelban, 2020-06-29:
Thanks.

I think it can be added as a separate module. 
I just never use this feature. So I do not prioritize this. But maybe someone will help with this module, or I just write it with time.

TymurP, 2020-06-29:
> Thanks.
> 
> I think it can be added as a separate module.
> I just never use this feature. So I do not prioritize this. But maybe someone will help with this module, or I just write it with time.

If I have some time i can write it as module, but first let me checkout your cool app 😄
Cheers🤙🏻

bviefhues, 2020-09-14:
A great first step would be to just visualize fan speeds.

MyAudioDNA, 2020-10-09:
I would agree with @bviefhues. If we can get just visualizing fan speed first, that would be great.

exelban, 2020-11-01:
Hi.

I add the Fans module in the [new](https://github.com/exelban/stats/releases/tag/v2.4.0) version.

closed 2020-11-01 by exelban

MyAudioDNA, 2020-11-01:
Thank you so much! That's awesome! Unfortunately, it isn't working for me.

When I turn fan monitoring on, I just get a white line in status bar and as soon as I click away, even the white line disappears. It looks exactly how the Sensors monitor looks if you turn it on without actually turning on any sensors.

[image]

I'm on a 2019 16" Macbook Pro and my OS and software are all up to date.

exelban, 2020-11-01:
> Thank you so much! That's awesome! Unfortunately, it isn't working for me.
>
>
>
> When I turn fan monitoring on, I just get a white line in status bar and as soon as I click away, even the white line [...]

Please open a new issue for this. It looks like some bug.

MyAudioDNA, 2020-11-01:
> Please open a new issue for this. It looks like some bug.

I have opened issue #135 for this bug. Thank you!