exelban/stats #51: App not visible in the Menu Bar

nikola-zitko, 2020-07-01:
After accidentally disabling all modules I can no longer access the app to enable them. I have tried reinstalling the app but to no luck. App is running the background (visible in the Activity Monitor) but I don't see a way to enable any module. 

P.S. I am new to MacOS so maybe I just don't know how to do it, any help would be appreciated.

exelban, 2020-07-01:
@nikola-zitko just close the app in the Activity Monitor and open again. It must start with settings window.

exelban, 2020-07-01:
Yeap, but there must be a way to open the application when you disable all modules.

nikola-zitko, 2020-07-01:
No settings window opens when launching the app.

exelban, 2020-07-01:
Hmm, so only one option when you have enabled Sensors widget, but not selected any sensor.
Ok, you can reset app settings by: `defaults delete eu.exelban.Stats`.

And restart the app. I will fix this problem in a few days.

nikola-zitko, 2020-07-01:
This fixed it, thank you very much.

closed 2020-07-05 by exelban