exelban/stats #52: Sensors not showing correctly

LauraJaime8, 2020-07-02:
[image]

exelban, 2020-07-02:
Hi. Thanks for the report.
Really strange behavior. As I see it's a dark mode. Does this appear when the app starts, or after click to the widget?

LauraJaime8, 2020-07-02:
This appear when select in app to show sensor in menu bar. Appear that whitout make a click in there.

exelban, 2020-07-15:
Unfortunately, I cannot reproduce this bug. It looks like something with macOS. In this widget, there are no background colors. So I have no idea, why the widget has a white background(
I tried it with macOS 10.14 and 10.15.

exelban, 2020-08-16:
Closed due to inactivity

closed 2020-08-16 by exelban