exelban/stats #53: [Feature Request] Selection of network interface

wernerfred, 2020-07-06:
Using the tool for a month now and i'm really happy about it. Just one little thing is bothering me: When connected to a docking station the network diagram only shows the WLAN interface. I would really appreciate if the tool would allow the manual selection of the interface of which the stats are shown. Another enhancement would be to be able to configure an autoselection of the used interface (automatically switching from wlan to lan if plugged in).

Any way i can help?

exelban, 2020-07-06:
@wernerfred Hi. Thanks for your feedback.

Yeap, it's possible. I will add this option with time.

exelban, 2020-07-08:
I add this option in a [new](https://github.com/exelban/stats/releases/tag/v2.1.7) version. Now you can specify what interface to show.

closed 2020-07-08 by exelban