exelban/stats #55: Feature Request: Touchbar Stats

Panther368, 2020-07-08:
Hey there,

I would like to ask if it would be a big deal to add support to show Stats on the Touchbar of an Macbook?
It would be sometimes awesome to have them shown in fullscreen applications.

P.s. I love your commitment with this application! :-)

Best regards
Pantehr368

exelban, 2020-07-08:
Hi.

Thanks for your feedback. It's possible, but I don't have enough time to implement this in the near future. But it's a nice feature, so I think I will add it with time. Thanks.

jcconca, 2020-10-26:
I already having a look into this.

https://github.com/jcconca/stats/tree/feature/add_touchbar_support

marcelochaves95, 2020-11-14:
Hi @exelban!
I have a MacBook with Touch Bar and I'm an iOS/macOS developer.
Can I help you with the implementation of this feature?

exelban, 2020-11-14:
@marcelochaves95 Hi. I don't even know. I never develop anything for the touch bar.
Even more, I'm not using the touch bar (only the escape button). So there is a lot of freedom for this feature. 
I only want to keep it more similar to the menu bar. And make it in the simplest way it's possible)

Maybe it possible to use the same widget as in the menu bar? It will be the best solution I think.

marcelochaves95, 2020-11-14:
@exelban is this the menu bar?

[image]

exelban, 2020-11-14:
Nope. [This](https://support.apple.com/en-gb/guide/mac-help/mchlp1446/mac) one.

marcelochaves95, 2020-11-14:
@exelban perfect. I will implement how is the menu bar.

Military-Doctor, 2021-01-15:
Working in progress: #291

exelban, 2021-04-04:
Hi. I'm not sure that this feature will be implemented.
I have no time for it. And I think that having monitoring in the touch bar is a good idea. The menu bar is good enough for it.

closed 2021-04-04 by exelban