exelban/stats #56: Upgrade 2.1.6 to 2.1.7 fails

bviefhues, 2020-07-09:
App-supported Stats upgrade 2.1.6 to 2.1.7 fails. Clicking "Check for new version" shows the new version dialog, plus the respective notification. However the install fails. 

The DMG file is downloaded, but the Finder window does not show, nor is there an upgrade started.

Opening the DMG file in finder and moving Stats to Applications works.

wernerfred, 2020-07-09:
Just updated right now. There was the typical "there is a new version available" screen. It downloaded and instead of showing the screen where you have to drag the stats icon to applications (like previously versions) now the notification slides in and tells me that 2.1.7 is installed (you need to allow stats to send notifications). Checking for updates it tells me i'm on 2.1.7.

So at least for me it worked. Maybe it did for you too but bc of turned off notifications you did not realize it?

bviefhues, 2020-07-09:
Nope. I have notifications turned on. I got the "New version is available" notification after clicking "check for updates" and I got the "2.1.7 is installed" notification after installing manually.

exelban, 2020-07-09:
It looks like it's related with notification.

In the latest version, I rewrite the updating mechanism. So now you don't need to move the App from dmg to Application. It's moving automatically now. But it looks like an application cannot be started if notification policy is not defined (accept or reject). I hope it's only the problem with the first update. I tried a few times to install the previous version (v2.1.6), and after update to v2.1.7 without any troubles.

bviefhues, 2020-07-09:
I will report here on how the next update behaves.

exelban, 2020-07-09:
Thanks. I will deploy a new version today, because of a bug in the network module.

exelban, 2020-07-09:
I already deploy a new version with a small fix. Could you try to update and check if there is no problem with updates?

bviefhues, 2020-07-09:
Same behavor on my machine: 
1) Open preference window
2) Click "Check for updates"
3) Stats charts stop being displayed in the menu bar (app is closed, I guess)
4) DMG is downloaded to Downloads Folder
Then nothing.

Manually restarting Stats: is still on 2.1.7 and a "New version available" notification shows

Clicking "Install" on the notification again closes Stats and downloads the DMG.

I can update manually.

exelban, 2020-07-09:
Hmm, I tried on a few machines.
I need to debug why it's not installing automatically. Could you check if `ls /tmp/Stats` exist?

exelban, 2020-07-09:
Please, check also `cat $TMPDIR/updater.sh`.

exelban, 2020-07-09:
And what macos version do you have?

bviefhues, 2020-07-09:
`cat $TMPDIR/updater.sh
cat: /var/folders/8r/j57v5lss6cg0bk4gl1tn_yl40000gn/T//updater.sh: No such file or directory`

Mojave 10.14.6

exelban, 2020-07-09:
hmm, and what output of `ls /tmp/Stats`?

bviefhues, 2020-07-09:
`/tmp/Stats`is empty.

exelban, 2020-07-09:
So it can be the problem with dmg mounting in 10.14.6. All my machines have 10.15.*.
Could you maybe help a little bit more? 

`/usr/bin/hdiutil attach "$DMG_PATH" -mountpoint "/tmp/Stats" -noverify -nobrowse -noautoopen`
Replace `$DMG_PATH` with the path to the latest dmg file. It just mounts the dmg under directory `/tmp/Stats`.

If it throws an error, I will change the `/tmp/Stats` to another directory (maybe permission problems, or `/tmp` does not exist).

bviefhues, 2020-07-09:
`/tmp/Stats`exists, is owned by me and is writeable `drwx------`

The `hdiutil`command returns `hdiutil: attach failed - Ressource ist belegt` - Resource is busy.

exelban, 2020-07-09:
`/tmp/Stats/` exist, because I creating this folder before mounting a dmg to it. So it looks like some problem with `hdiutil`.
But why the resource is busy... Maybe you have mounted this dmg?

bviefhues, 2020-07-09:
Yes, was indeed mounted. Now it works. 

`hidutil`output:
`/dev/disk2          	GUID_partition_scheme          	
/dev/disk2s1        	Apple_HFS                      	/private/tmp/Stats`

`ls /tmp/Stats`
`Applications	Stats.app`

exelban, 2020-07-09:
It's possible that you have mounted the new dmg in the update process?
Maybe you noticed that the app closed and open a downloaded dmg?

bviefhues, 2020-07-09:
I did mount the DMG manually, via Finder, for the latest upgrade. For next release, I will observe the mount behavor and report back here.

exelban, 2020-07-09:
I know that I'm asking too much)

But could you check one more thing? Download the [v2.1.7](https://github.com/exelban/stats/releases/tag/v2.1.7) manually, install it. And try to update it. If it will fail, I will install a virtual 10.14.* and debug it.

bviefhues, 2020-07-09:
(You are not asking too much, I am happy to help.)

Here is what I did:
- Quit Stats
- Move all Stats DMGs in Downloads to paper basket
- Deleted Stats in /Applications (moved to paper basket)
- Downloaded 2.1.7 Stats.dmg
- Manually installed 2.1.7 (opened DMG, dragged to Applications)
- Launched Stats.app
- Got two notifications: 1) New version available 2) Successfully updated
- Closed the "Successfully updated" notification
- Clicked "Install" on "New version available" notification
- Stats terminated itself
- And did not restart
- There is a new/2nd Stats-1.dmg in Downloads
- Re-started manually 
- Got New version available notification

Same behavour when clicking "Install" again.

exelban, 2020-07-09:
Yeap, thanks. It looks like there is some problem with the update script. I will debug it tonight.

exelban, 2020-07-09:
Ok. I install the macOS 10.14.4 on VMware. It looks like there is a problem only with macOS Mojave (10.14.*). 
Also, I figured out where exactly the bug hide. And I totally do not understand why it's not working.

But after a few attempts, I found a solution. And it works on VM. But I'm not sure if it will work for real mac (should). I will deploy a new version with this fix. 

And if you will have a few minutes, give me to know. After this fix, I can deploy a test version. Just to ensure that the update mechanism works.

exelban, 2020-07-09:
There is a [new](https://github.com/exelban/stats/releases/tag/v2.1.9) version. Unfortunately, v2.1.7 and v2.1.8 have broken update mechanism (only in macOS Mojave). So you need to update manually.

bviefhues, 2020-07-09:
Does the fix need 2.1.9 installed for the upgrade to work? I tested with 2.1.7 installed and tried the upgrade (via "Install" from notification) and that failed with same behavour.

exelban, 2020-07-09:
Unfortunately, it's not over. Now another problem arrear. MacOS block dmg mount. So it looks like only users with disabled SIP can update the application(

exelban, 2020-07-09:
At least I know how I can fix this.

exelban, 2020-07-09:
Ok, I hope I fix all problems. I can update the application on 10.14 and 10.15. Without disabled SIP.
And yes. I will be sure only after the next update. But I hope it works)

closed 2020-07-09 by exelban