exelban/stats #58: App Release / Update frequency

vviikk, 2020-07-12:
Love this app. Any chance we can have a dropdown in settings on how frequently to check for updates? I am getting bombarded by so many releases. This is a GOOD thing that the app is active, but would love it if there's an option to check maybe once a week.

[image]

exelban, 2020-07-12:
Wow. Do you receive this notification at once?

vviikk, 2020-07-13:
This was overnight... maybe over 24 hours?

exelban, 2020-07-13:
Ohh, I see. I have a cron job that checks for updates every 12 hours. And it does not check if notification dismissed or not. I will add the option to select an update interval check. But I still must check if notification dismissed or no.

exelban, 2020-07-14:
I add options to select how often app will be check for the new version:
- at start
- once per day
- once per week
- once per month
- never

This feature will be available in the next release.

exelban, 2020-07-19:
Available in the [new](https://github.com/exelban/stats/releases) version.

closed 2020-07-19 by exelban

exelban, 2020-07-19:
I will also try to deploy a new version once per week (on Sunday). There were soo many chaotic deployments because some time ago I rewrote the app from scratch. And I just fixing the bugs. Now, it looks like the app is stable and I can move to the normal releases intervals.