exelban/stats #59: Feature Request: Different battery icons when plugged/unplugged to power

russoz, 2020-07-14:
Hi, 

Thanks for putting this awesome app together. It is really helpful on my daily work.

It would be great to have some way of visually telling, just by looking at the menu bar whether AC power is on or not. 

Well, I can obviously see the computer itself, but I had an issue a while ago on the previous laptop, when some of the USB ports were not working at all, so even though I had the cable plugged in, the battery was not charging. And on a different occasion, it was - apparently - a software problem: even though the cable was plugged, MacOS "thought" it wasn't, so eventually it put the computer to sleep "believing" the battery was drained. After a reboot, everything was normal.

Hence, it would be nice-to-have some indication of the AC Power status (without having to click on it).

Thanks once again!

Cheers,
Alex

exelban, 2020-07-14:
Hi. Thanks for your feedback.

There is an indication for AC Power status. It's not great, but still. If a laptop has a connected power cable (and its charge), the battery frame changes color to green.

In the feature, I will redraw the battery icon, and make ac power more visible (maybe like in system icon).

russoz, 2020-07-17:
Hi @exelban thanks for your prompt response. I had missed that!!! Now that you wrote it I realized it was there all along. Well, thanks for considering the suggestion though.

exelban, 2020-07-19:
I finally finish the battery widget and now it works in the same way as system one. There is no green frame. But normally it shows lightning when the power cable is connected. Also, improve reaction when the power cable is connected.

Available in the [new](https://github.com/exelban/stats/releases) version.

closed 2020-07-19 by exelban