exelban/stats #62: colour coding for "based on utilization" settings

bluegrover, 2020-07-19:
Hi, 
when selecting a colour coding for "based on utilization" the colour seems to ba orange per default (if levels are low/ normal)
My assumption would be the colour to be green-ish if the utilization is low or normal or CPU, RAM, SSD and should then become orange or red-ish if higher or critical levels are reached. (I think it was similar to this in one of the more early versions of stats). 
Currently the change seems to be from orange to red when high/critical levels are reached, which I don't really find logical. Green to red/orange for me would seem more intuitive.
cheers
bluegrover

exelban, 2020-07-19:
For now, there are 3 colors for utilization:
- 0% - 60%: system accent color
- 60% - 80%: orange color
- 80% - 100% red color

bluegrover, 2020-07-19:
Hi Excelban,

understood now.
My macos Highlight colour/system accent colour is set to orange. That's why for me then 0-80% range will look orange.
Would it be a big deal to change the 0-60% area colour from system accent to green (just to follow the traffic-light scheme from red-orange-green for "based on utilization")?  At least I would pretty much prefer that over system accent clour :-))
Because system accent colour is a separte selection you can take in the drop-down settings...
Thanks very much anyway for your quick response.
Best, Bluegrover

exelban, 2020-07-19:
Previously it was a blue color. I will revert it to the next release. So utilization colors will look like:
- 0% - 60%: blue color
- 60% - 80%: orange color
- 80% - 100% red color

bluegrover, 2020-07-19:
Hi Exelban,
that is most kind.
Blue is absolutely fine, but I think the colour for "second green" is even nicer ;-)
cheers and thanks very much, 
BG

exelban, 2020-07-20:
Update in the [new](https://github.com/exelban/stats/releases/tag/v2.1.13) version.

closed 2020-07-20 by exelban

bluegrover, 2020-07-20:
Excellent!
Thank you
Just installed