exelban/stats #64: Battery triggers system notification below 10% even when charging

russoz, 2020-07-25:
Hi,

As it seems (haven't done much testing on that), whenever the battery level is below 10%, every time the level changes by 1%, a system notification is triggered. Skipping the part that this might be configurable, I had the computer battery down to 4% then I plugged it back in: it triggered the notification when charged to 5%, then 6%, 7%, etc.. until 10%. 

These notifications are very useful when discharging, but I believe it doesn't make sense to trigger these notifications when charging.

exelban, 2020-07-25:
Hi, thanks for the report.

Did you use the latest version v2.1.13?

russoz, 2020-07-25:
hi @exelban I missed that one last update, I had been running 2.1.12. I have just updated and will repeat the test soon (probably tomorrow, it's late around here ;-))

exelban, 2020-07-26:
I noticed this bug in some moment, and fix it in the latest version.

russoz, 2020-07-26:
Yup, I tested it again last night with the latest version and it was fixed.
Thanks!

closed 2020-07-26 by exelban