exelban/stats #67: Celsius Temperatures Shown as Fahrenheit

adamddh, 2020-08-10:
When showing temperatures in the power section, temperatures are marked incorrectly. Solutions include:
1) Changing the symbol to a C (thus making the measurements accurate)
2) Converting to Imperial Units (thus making the measurements accurate)
[image]

exelban, 2020-08-10:
Hi. 
Thanks for reporting this. 

As I understand, it shows value in C, but units in F. I will fix it for the next release.

exelban, 2020-08-16:
Fix it in the [last](https://github.com/exelban/stats/releases/tag/v2.2.2) release.

closed 2020-08-16 by exelban