exelban/stats #68: Network speed by process showing 0 KB/s

mariusv, 2020-08-10:
**Describe the bug**
When clicking on the networking pop-up most of the time the network speed is not showing up even if those processes have traffic

[image]

**Details:**
 - Device: MacBook Pro 13-inch, 2019
 - MacOS 10.14.6
 - Application version 2.2.1

exelban, 2020-08-10:
Hi. Thanks for reporting.

Does it always shows 0 value?

mariusv, 2020-08-10:
Nope, every once in a while shows 3KB or something

exelban, 2020-08-10:
Thanks.

I read a `nettop` with 1 sec interval and calculating the difference. It's a not good approach to read an IO. But something. Could you maybe try this command: 
```
/usr/bin/nettop -P -dk time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch
```
And check if show better stats?

mariusv, 2020-08-10:
you are right, the stats are kinda the same. Thank you

exelban, 2020-08-10:
Thanks for this. 
I also noticed some strange behavior of these stats. But I cannot get a better read. It requires root privileges. But I don't want to ask users these privileges.

mariusv, 2020-08-10:
Yeah, was looking at little snitch (I'm using it to block connections) and that one has better stats and that's how I noticed this :-)
I will close this issue now as there is nothing to action on. 

Thank you very much for the hard work and the quick reply

closed 2020-08-10 by mariusv