exelban/stats #70: Charts should have axis scale info

kim0, 2020-08-16:
Thanks for your recent work adding charts for network info, and improving the visuals everywhere. It looks amazing :) However, it's still hard from the network graph to tell what that spike means, was it 1Mbps or 100Mbps ..etc. I think for graphs like Memory and Network, some kind of numbers on the y-axis are needed. At least what the top value of the graph means. But ideally a couple of more data points (say 33%, 66%, 100%). The CPU might not need it, bec the max is always 100%.

Thanks!

exelban, 2020-08-16:
Hi. 

I will add it with time. Or maybe someone will help with it. Because it's not critical for me. So it can take some time.

kim0, 2020-08-16:
Absolutely, thanks a lot for the time you invest in this app. Really appreciating your time and effort. Cheers mate

exelban, 2020-08-18:
Hi. 

Ok, I don't like this chart at all.
So I redraw it. Also, I add the information about the maximum visible value in the chart.

[image]

It will be available with the next release.

kim0, 2020-08-18:
Great stuff! Your work is awesome. Thanks a lot.

If I may suggest adding a light horizontal line at the 50% mark, would make
this much easier to read. Although it's already a huge improvement! 👍