exelban/stats #74: A

trung753dfx, 2020-08-25:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - MacOS [e.g. 10.15.5]
 - Application version [e.g. 2.1.11]

exelban, 2020-08-25:
Hi. Thanks for reporting. Could you share a crash report?

It's available in Console.app or in ~/Library/Logs/DiagnosticReports.

trung753dfx, 2020-08-25:
Here is the [Crash Report](https://github.com/exelban/stats/files/5122036/Stats_2020-08-25-142730_Trungs-MacBook-Pro.txt).

exelban, 2020-08-25:
@trung753dfx can I send you a build for a test?
Because I don't understand clearly from report what goes wrong. It's a problem with the reader (collecting a GPU stats) or maybe some UI bug.

exelban, 2020-08-25:
I would appreciate if someone with similar bug checks if this version works.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5126124/Stats.dmg.zip)

trung753dfx, 2020-08-26:
I have install your new build and it still crash, seem the application can't find iGPU information and crash.
If you still need crash report, tell me.
[image]

exelban, 2020-08-26:
Thanks. Is not showing any GPU on the list because the reader is turned off. I will make a new test build with another change. Could you say if you have any graphics in the `Stats` tab?

exelban, 2020-08-26:
Try this one please. It must prevent crashing. But I think you will not see any GPU.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5128344/Stats.dmg.zip)

trung753dfx, 2020-08-26:
The new build is not crashing anymore. 
I tick the GPU and it show up GPU icon on my menu bar but it isn't monitoring anything.
[image]

>Could you say if you have any graphics in the `Stats` tab?
[image]
Nothing happen when I click it. No blank drop down or something.

exelban, 2020-08-26:
In the settings, there is information about the device. And there is a info about graphics. Did it show something in the GPU section?
[image]

trung753dfx, 2020-08-26:
Sorry, I misunderstood your question.
Here is my Stats tab
[image]

exelban, 2020-08-26:
Thanks. At least now I know that there this GPU detected. It looks like app cannot read needed properties of this GPU. 
Could you maybe run this command: `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt` and send `output.txt` file.

This command returns a list of GPUs available on mac with their specs. The same command using reader to fetch the GPUs in the add.

trung753dfx, 2020-08-26:
Here you go [output.txt](https://github.com/exelban/stats/files/5128831/output.txt)

exelban, 2020-08-26:
Thanks, it looks like everything is ok. So it must be a problem with some parameters. I prepare a new build that has enabled logs to file. I would ask you to run this build. It does not have any fixes. But it will write a logs to `Documents/log.txt`. It just writes all steps from the GPU reader. I will be appreciated if you can send me a log. 

[Stats.dmg.zip](https://github.com/exelban/stats/files/5128969/Stats.dmg.zip)

trung753dfx, 2020-08-26:
[log.txt](https://github.com/exelban/stats/files/5129028/log.txt)

exelban, 2020-08-26:
Thanks, it looks like it cannot match device from `IOPCIDevice` and `IOAccelerator`.

exelban, 2020-08-26:
Could you maybe run this one and send the logs (same as previous)?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5129076/Stats.dmg.zip)

trung753dfx, 2020-08-26:
What a long log 
[log.txt](https://github.com/exelban/stats/files/5129160/log.txt)

exelban, 2020-08-26:
Ohhhh. I see. Capitalizing bug)
```
pciMatch: 0x0D268086
accMatch: 0x0d268086 0x0a268086 0x0a2e8086 0x0d228086 0x04128086
```

0x0D268086 != 0x0d268086

exelban, 2020-08-26:
If there is no another bugs, this one must works.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5129193/Stats.dmg.zip)

trung753dfx, 2020-08-26:
Cheer, GPU monitoring is now working. Every tick box is working and the drop down show my iGPU.
[image]
Do you need I send you the new log file?

exelban, 2020-08-26:
Nope. You can remove the logs. Today I will deploy a new version of the app with this fix. Thanks for your help and time.

exelban, 2020-08-26:
Fixed in the [new](https://github.com/exelban/stats/releases/tag/v2.3.1) version.

closed 2020-08-26 by exelban