exelban/stats #75: no more Icons shown

warmup72, 2020-08-26:
After using Stats for som weeks I couldn't chance my prefs anymore.
The CPU-Icon is shown I the statusbar, but Activation of other Views (e.g. Network or RAM) did not work.
New start of stats and my map were not usefull

MBP 16" . 10.15.6
Stats 2.3.0

exelban, 2020-08-26:
Hi.

What do you mean?
Did you see only CPU stats in the menubar? Or all shows?

warmup72, 2020-08-26:
Excuse my bad English. I only displayed the CPU usage in the status bar. Well after 2 weeks I can't activate any other ads. For example, if I set the network switch to, nothing is displayed in the system bar except the CPU load.

exelban, 2020-08-26:
So, you have only CPU module active. And some time ago you disable other modules. And now you cannot activate disabled modules when you triggering the toggle?

warmup72, 2020-08-26:
yes correct

exelban, 2020-08-26:
Yeap, I suppose I know where is the problem. I will debug it today and return with results.

Thanks for reporting this.

exelban, 2020-08-26:
Fix in the [new](https://github.com/exelban/stats/releases/tag/v2.3.1) version.

closed 2020-08-26 by exelban

warmup72, 2020-08-27:
Wow magic, fast work!
Thx works fine!