exelban/stats #79: Feature Request: More Top Processes in CPU module

warmup72, 2020-08-31:
I think it would be great if you could give the user a choice of how many TopProcesses he would like to see displayed in the CPU module

exelban, 2020-08-31:
Yeap, thanks. I will add this feature.

warmup72, 2020-09-22:
If You add something in the cpu modul I think It would be cool if you separate user- and system-usage in to different figures on the menu bar

exelban, 2020-09-23:
Thanks for this proposal. I think it's not related to this issue (more processes in top list). Will be great to have it in a separate issue. But, in this case I think I will not add this feature. It's too light level setting. And will provide a big complication for users who just want to see CPU usage. Also, I don't see any good solution to how it can be implemented (from UI/UX part).

exelban, 2020-11-01:
Implemented in the [last](https://github.com/exelban/stats/releases/tag/v2.4.0) version.

closed 2020-11-01 by exelban