exelban/stats #81: stats help

randy9999, 2020-08-31:
Is this normal?
[image]

I am getting really high cpu temps if I am reading this right.  Please look at the included img.  Is this a concern for me or is your stats program the concern?

thanks

exelban, 2020-09-01:
Hi.

Strange. I suppose you are confused about CPU 1 and CPU 2 temperature?
It's a second time when I see high values on these sensors. 

What mac do you have?

randy9999, 2020-09-01:
Thanks for getting back to me.

I have an i9 27-inch iMac, 2019.  With 24 GB of memory.  It also has Radeon
Pro 580X 8 GB graphics.

exelban, 2020-09-01:
Stange situation. Because I just read the value from SMC. It looks like macOS reports the wrong value from sensors. Or maybe I wrong convert the value. But still the another sensors looks ok.

randy9999, 2020-09-01:
Yes, I agree that this reading is a little strange...that's why I reached
out to you.  I forgot to mention that I also am running macos 10.15.6., the
latest version.  If that makes any difference.

exelban, 2020-09-01:
Did you ever use another monitor? If yes, it shows the same strange values on these sensors?

randy9999, 2020-09-01:
No, I do not use any other monitor.  Just the imac monitor.

exelban, 2020-09-02:
There is one problem. MacOS does not have an API to fetch a sensor value. So you cannot check throw terminal the value. There are only specific tools for this. Maybe you will have some time to install this tool and check if sensors returning a proper value?

randy9999, 2020-09-02:
Ahh, what are you talking about?  I did not get anything other than this
email...no attachment, nothing.  Nothing in my trash either.

exelban, 2020-09-02:
Maybe [reset the SMC](https://support.apple.com/en-us/HT201295) will help. Could you maybe try it?
I review all possible options. There is no way that my app process the value in the wrong way. It just read the value from SMC and show it.

randy9999, 2020-09-02:
> value in the wrong way. It just read the value from SMC and shows it.

Ok, I did the reset SMC routine.  Here are the results...

[image: Screen Shot 2020-09-02 at 3.16.02 PM.png]

To me they look just about the same as the first results.  So I am back to
the same question.  Are these results something I should be concerned about
on my computer?  Or are these results something that you should be
concerned about on your program?  i.e. Then this is nothing for me to worry
about.

exelban, 2020-09-03:
It looks like just macOS report the wrong values. It's not critical. And it's not a bug of this app.

exelban, 2020-09-06:
Ok, I update the CPU sensor reader. And remove temperature sensors with anomaly hight value from the list.

closed 2020-09-06 by exelban