exelban/stats #83: Network icon is shown but not activated in the options

NoTrace-Kai, 2020-09-04:
Hallo ...
When my MacBook starts the Network icon is shown in the menubar,
but is is not activated in the options of stats.
macOS Catalina 10.15.6
Thanks for a little help.
CU Kai

exelban, 2020-09-04:
Hi, thanks for reporting. I will check this.

exelban, 2020-09-04:
Can you provide a little bit more information?

Did you have the latest version of Stats installed?
Did you try to turn on -> turn off? Did it still appear after this procedure?

NoTrace-Kai, 2020-09-04:
Hi ...

yes, I used the latest version of Stats and yes after a restart of my MacBook Air 2020 the Network Icon
was again there.

CU

exelban, 2020-09-04:
Thanks.

So, it shows the network widget but the values are not updating. Always showing 0/0?

NoTrace-Kai, 2020-09-05:
I tried 2 reboots now ... no network icon is shown ... I am not sure if the showing 0/0 or the live traffic.

exelban, 2020-09-05:
Does reboot help? And now network is not showing?

NoTrace-Kai, 2020-09-05:
Yes, for the moment. But as I wrote the "bug" rebooted the MacBook also ... ?!

exelban, 2020-09-05:
Strange. If it appears again, please give me to know. Also, notice if the values are changing.

exelban, 2020-09-05:
Ok, it looks like I figured out where is the problem. Trying to fix it.

NoTrace-Kai, 2020-09-05:
Great!

exelban, 2020-09-05:
A short explanation of why it's happening.

Network module a little bit different from the others. It has a listener when the connection state is changed (wifi on/off). And this listener is set up when app is initializing. So, module can be disabled, but if the network state changes, it shows the widget.

There were 2 bugs: 
- showing the widget when module is enabled
- listening for network state change when the module was disabled

Fix for this will be tomorrow in the new release.

Thanks for reporting.

exelban, 2020-09-06:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.3) release.

closed 2020-09-06 by exelban