exelban/stats #84: Empty spaces on disabled categories on macOS Big Sur

Uffje, 2020-09-08:
[image]

I disabled all categories except RAM and CPU, but the disabled one are still "visible" in the menubar.

exelban, 2020-09-09:
Hi. Thanks for reporting.

For now, the app is not compatible with macOS Big Sur. There can be a lot more bugs because the app is not ready for the new version of macOS.

exelban, 2020-09-09:
I tried macOS big sur. It looks like they add spacing to each element in the menu bar. 

Now when you disabling the widget, it sets the wight to 0. In the previous version of macOS, there were no paddings. So it works perfectly. But now, when padding exists, it's not hiding the widget.

It set the width to 0, and not removing the element from the menu bar. Because there is only one option to keep the widget order. Otherwise, the ordering will have strange behavior. But for now, there is only one option. I hope Apple will add some API for this, or they will remove the paddings if the widget width is 0.

exelban, 2020-10-04:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.6) release.

closed 2020-10-04 by exelban