exelban/stats #86: Network stat shows zero bytes for download value

hoondi, 2020-09-16:
It appears that if any non NIC (and therefore non Apple kext) is used for wired networking, the download value is always zero.
I am using an ATTO ThunderStream 10GB copper ethernet adapter connected via Thunderbolt to a MacPro.

[image]

If I elect to use WiFi, then everything back to normal as working as expected.

This is also observed if with Menumeters and Apple's Activity viewer and so Apple have changed something in a delta release approximately around 10.15.3 from memory.

I'm aware that Apple want to minimise developers writing kexts and so am wondering if the above is a result of Apple preparing for this? 

Regards,
Raoul.

exelban, 2020-09-23:
Hi. 

I tried to debug was going on with thunderbolt ethernet. It looks like the problem is somewhere in the system. I don't know exactly where. In the apple kext or in the ethernet device. Also, I can't fix this. I just read the stats from the system about a specific interface.

exelban, 2020-10-04:
Sorry but it looks like the problem is not on the app side but in macOS. I can't fix it.

closed 2020-10-04 by exelban