exelban/stats #87: Option to show speeds as either Bytes or bits.

adamddh, 2020-09-17:
Option for read/write and/or upload/download speeds to be in *B/s OR *bit/s.

exelban, 2020-09-25:
Hi. 

Why do you need to have bites in the speed? You will never (maybe in very specific cases) saw the value in bites. In 99.9% it will be >1 KB/s.

adamddh, 2020-09-25:
Having the option of changing the speed of bits or bytes give ease of mind that the user can choose the standard (bits) or choose bytes for those uncomfortable/unfamiliar with bits.

exelban, 2020-09-25:
Ohhh, sorry. Now I see. But these stats are not the speed. It's the transferred size in the last second.

adamddh, 2020-09-28:
While that does fall under the umbrella of speed, having the transfer size in either format is the same enhancement.

exelban, 2020-10-04:
Added in the [last](https://github.com/exelban/stats/releases/tag/v2.3.6) release.

closed 2020-10-04 by exelban