exelban/stats #90: Show single large sensor value if only one sensor is selected

mobidiv, 2020-09-22:
If I select only 1 sensor in the Sensors section, it appears small in the upper index remaining a blank space for the other sensor value. It would be great to show it large fitting the whole space.

mobidiv, 2020-09-22:
Just figured it out: a new single value widget will fit better. I'd like to spread a few single sensor values in the menu bar.

exelban, 2020-09-23:
Hi. What do you propose?
Showing the value in the center with the biggest font size when there is only one value?

mobidiv, 2020-09-23:
Hi!

> Showing the value in the center with the biggest font size when there is only one value?
Correct.

exelban, 2020-09-23:
Thanks. I will implement this in the next release.

exelban, 2020-09-27:
Added in the [new](https://github.com/exelban/stats/releases/tag/v2.3.5) version.

closed 2020-09-27 by exelban