exelban/stats #92: Network Up/Down information should have dynamic spacing

codebytere, 2020-09-24:
**Describe the bug**

The Network `NSMenuItem` is intended to account for what looks to be a threeish-digit max: 
[image]

But does not shrink or expand to account for the current numeric value, leaving perhaps too much whitespace at lower values:
[image]

It'd be awesome if the dots/characters/etc accounted for this so as not to create quite as much whitespace.

**Details:**
 - Macbook Pro 2019
 - MacOS Catalina 10.15.6
 - Stats v2.3.3

exelban, 2020-09-24:
Hi.

TL;DR static size of this widget allows ensuring the static position of elements in the menu bar.

There is one problem with this approach. Network value is too dynamic. I mean that value (upload/download) can change each second. And if widget size will be dynamic (depends on text size), the widget will be jumping. For example, if youtube (or some stream) is opened, the download value can jump from 0 KB/s to 10.3 MB/s. And if the network widget is not first in a row, your all menu bar will be jumping when the network is active (because the width of the widget change).

codebytere, 2020-09-27:
Thanks for the background! that does make a lot of sense, and either way it's a very small deal in the big scheme of things :) Thanks for your work on this app, and feel free to close this issue out as you see fit.

closed 2020-09-27 by exelban