exelban/stats #93: Enhancement: Possible to list all GPUs

hoondi, 2020-09-25:
MacPro 6,1 comes with multiple GPUs as do many MBPro laptops.

Just wondering if a future version could show both GPUs as I've also wondered how much work multiple GPUs actually do with FCPro etc etc.

luv ya work!

exelban, 2020-09-25:
Hi. What graphics are missing?

exelban, 2020-09-25:
Hi. 
Nope, you must select one GPU to show. In 90% of the time, only one GPU is primary. Even when you have enabled dedicated GPU, integrated work as a helper in some calculation. I don't see any reason to see a few GPUs in the menu bar.
Only when you have a few dedicated GPUs and computing something on both of them (some NN) there is some sense to see multiply in the menu bar.

closed 2020-09-25 by exelban

exelban, 2020-09-25:
I have from time to time a few dedicated GPUs. So for me, this feature can be helpful. But there is no option to implement 2 widgets per module. It requires rewriting a big part of the application. Maybe in the feature, I will add this option. And then it can be possible to implement this feature.