exelban/stats #94: network values not showing correct value

hoondi, 2020-09-25:
Just updated to macOS 10.15.7 on MacPro 6,1.

I wish I had an upload speed of 204MB/s to the internet.
[image]
This value is frozen.
Have quit Stats.app to no avail.
changing Stats.app to monitor WiFi works, but when I go back to Ethernet, it goes back to 204MB/sec.
interfaces active are ATTO 10GB Copper via Thunderbolt and Wifi:
`en11: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 9000
	options=46f<RXCSUM,TXCSUM,VLAN_MTU,VLAN_HWTAGGING,TSO4,TSO6,CHANNEL_IO>
	ether 00:10:86:82:2d:ee 
	inet6 fe80::86b:b863:1d48:61ee%en11 prefixlen 64 secured scopeid 0x8 
	inet 192.168.1.202 netmask 0xffffff00 broadcast 192.168.1.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect <full-duplex,flow-control>
	status: active
en2: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	ether 18:af:61:b9:1f:c4 
	inet 192.168.1.235 netmask 0xffffff00 broadcast 192.168.1.255
	media: autoselect
	status: active`

The upload value has just gone up to 205MB/s so looks to be an aggregate value instead of current value?

Given that Apple's Activity Viewer is accurately reporting my upload speed, I figured I'd report it.

Cheers,

hoondi, 2020-09-25:
just reset all network settings by deleting applicable plist files from /Library/Preferences/SystemConfiguration/ and so all my interface names have been reset etc etc

I launched stats.app just now and it's constantly showing 813KB/s for upload…
[image]
Now it's at 1006KB/s
[image]

so something is not write with Stats.app.

Cheers

exelban, 2020-09-25:
Hi.

Nice research. It looks like the same problem as in the previous ticket (#86). As I say on that ticket. It's not rather the problem in the app but in the kext or device.

What model of thunderbolt network card do you use?
Maybe you have a few of them? If yes, did this problem exist on that device too?

hoondi, 2020-09-29:
it's an ATTO ThunderStream dual 10GB copper (ThunderLink NT2102) , in a Thunderbolt2 enclosure… Don't have two of them I'm afraid.. 

Looking under the hood, [PCI Authority](https://pci-ids.ucw.cz/read/PC/8086/1528 ) says it's an Intel Ethernet Controller 10-Gigabit X540-AT2, so nothing exotic.

I have Chelseo and other Intel 10GB NICs here, but no spare PCIe thunderbolt chassis to mount them in to test…

exelban, 2020-10-04:
Sorry but it looks like the problem is not on the app side but in macOS. I can't fix it.

closed 2020-10-04 by exelban