exelban/stats #95: Weird spacing in second (unfocused) menu bar

dpatterson-sf, 2020-09-29:
**Describe the bug**
This is pretty minor, but in a dual screen setup, the menu bar items move to a different location when the screen loses focus.

**Details:**
 - Device: Mac Pro 2013 (trash can)
 - MacOS 10.14.6
 - Application version 2.3.5

[image]
[image]

exelban, 2020-09-29:
Hi.

Thanks for the report. I will take a look at this strange behavior.

dpatterson-sf, 2020-09-30:
i just discovered that restarting the Stats app fixes this. so i guess it only happens when you remove some widgets, and it's super minor, and there's a workaround

locussss, 2020-09-30:
I have the same behaviour but on my main and only monitor.
And it won't change after a restart of the application, neither on a computer restart.

Mac mini (2018), 3 GHz 6-Core Intel Core i5, Intel UHD Graphics 630 1536 MB on macOS 10.15.5

exelban, 2020-09-30:
Yeap, I already debug this. The problem appears only when you disable the widget. It will be fixed in the next release (next Sunday).

exelban, 2020-10-04:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.6) release.

closed 2020-10-04 by exelban