exelban/stats #96: Several hardware items wrongly reported

bouscram, 2020-09-29:
Hi. I have a 2012 iMac with a NVIDIA GeForce GT 650M 512 Mo GPU and a 1 TB Fusion Drive
Here is what Stats reports for the configuration: 
[image]

There are several mistakes: the Intel GPU does not exist (to my knowledge), and only the SSD part of the Fusion drive is reported.

Also, I have remarked that, while I have 4 physical cores and no hyperthreading,  I need to select both cores and hyperthreading to see the 4 cores in the CPU widget:

[image]

Keep on going, your program is really promising (I particularly like the view on top processes). One thing I would really like is a graph widget for network traffic.

exelban, 2020-09-30:
Hi.

Your mac has 2 GPUs: Intel HD 4000 and NVIDIA GeForce GT 650M. Intel HD 4000 is integrated into CPU, but [it's not available to display](https://forums.macrumors.com/showthread.php?p=16920033#post16920033). 

The hard drive looks really strange. I suppose this `Apple SSD SM128E` it's a fusion drive SSD. At least [google helps me with this](https://support.apple.com/en-us/HT207584). It says that Fusion Drive is a 2 disk: SSD + HDD. In your case is SSD (128GB) and 1TB HDD. Unfortunately, I do not have any fusion drive to detect properly this type of disk. But if it shows as 2 drives (HDD + SDD), I cannot properly detect this disk at all.

I just read these parameters from the system. It's not a hardcoded list of devices.

The bug with CPU I can fix. It's related to the way how I read the activity of CPU. I receive the full list of available cores. From this perspective, the app does not know if it's a hyperthreaded core or no. When the hyperthreaded cores option is disabled, I just remove `i+1` core from the list. So in your case, the hyperthreaded option must be unavailable and turned on.

bouscram, 2020-09-30:
Thanks for your answers, knowing that there is an (almost) unused Intel GPU in my machine is really a surprise...

Regarding the HD, when I look at the stats themselves, the volumes (not the physical HD) are correctly seen, so this is not really a big issue. There is however a problem, because the Recovery volume is shown as having the same size and occupation as the main drive. I guess it should not be shown.

exelban, 2020-09-30:
Do you see the Recovery disk in the popup view?

bouscram, 2020-09-30:
Yes:
[image]

(AluMac is my main volume, Black is an external USB disk)

exelban, 2020-09-30:
Thanks, I will fix it.

exelban, 2020-10-04:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.6) release.

closed 2020-10-04 by exelban