exelban/stats #97: Feature request: new menubar widgets

bouscram, 2020-10-01:
Hi.
I have been a longtime user of MenuMeters (which is now back in semi-active development here: https://github.com/yujitach/MenuMeters), and while I really prefer the info windows of Stats (more info, nicer), I miss my favourite menubar widgets. Here is my usual setup with MenuMeters (particularly the CPU and network widgets):

[image]

(network on the left, 4-core CPU in the middle and memory on the right)

Can you add some of these in your development plans?

exelban, 2020-10-01:
Hi.

Can you tell a little bit more about these widgets?

bouscram, 2020-10-01:
I guess the best for you would be to install MenuMeters and see for yourself, but they are quite simple: the network widget shows 2 history graphs (with different options for the scale - including linear, square, cubic and logarithmic). The CPU widget has several option, but this one is just an history graph for each core. The width of the widgets is also customizable.

exelban, 2020-10-01:
Ok, I will take a look later today.

exelban, 2020-10-04:
I take a look at this app. I like the pie chart to visualize the ram. And I can implement these as a widget.

closed 2020-12-06 by exelban