exelban/stats #98: Empty space is created and stay display if click on on section or move/reorder icons in menu bar

Thmsdc, 2020-10-05:
**Describe the bug**
I did check the ticket #84 but seems a little bit different.

When Stats is launched it displays different stats in the menu bar without any space issue. From the moment one specific stat is clicked to be seen or icons moved or reorder (other apps and/or Stats) a space is created and stays unless Stats is closed and relaunched.

**Details:**
 - Device: Mac mini 2018
 - MacOS 10.15.17
 - Application version 2.3.6

[image]

[Screen Recording.zip](https://github.com/exelban/stats/files/5328131/Screen.Recording.zip)

exelban, 2020-10-05:
Hi. 
Thanks for the report. Could you check if there is no enabled but not visible some module? 
If not, try to enable/disable disabled widgets.

Thmsdc, 2020-10-05:
Hi.
Unfortunately it seems then all the stats are activated/disabled correctly. For instance if I activate my Network and Sensors which are disabled originally it will add them but the empty space still going to be there.

Made a record to highlight the issue and my tests.

[Screen Recording 2020-10-05 at 23.15.27.mov.zip](https://github.com/exelban/stats/files/5330132/Screen.Recording.2020-10-05.at.23.15.27.mov.zip)

exelban, 2020-10-06:
Thanks.
So even when all modules enabled there is this an empty space?

Thmsdc, 2020-10-06:
Yes I confirm than even with all modules turned on white space is still present. (on the right here)

[image]

exelban, 2020-10-06:
Hm, it's super strange. It looks like the empty space is a space for some module. But all modules enabled. So it can be some bug with battery module.

jaykovski, 2020-10-10:
> Hm, it's super strange. It looks like the empty space is a space for some module. But all modules enabled. So it can be some bug in the battery module.

Having same problem, all modules enabled, except battery, since my Mac doesn't have a battery and the option is not appearing in the app.

exelban, 2020-10-10:
Could you provide the model of mac?

jaykovski, 2020-10-10:
> Could you provide the model of mac?

27" iMac 2019 (iMac19,1)

exelban, 2020-10-10:
When you click on this empty space something happens?

jaykovski, 2020-10-10:
> When you click on this empty space something happens?

Only highlights the background when you click on it.

exelban, 2020-10-10:
Hmm, strange behavior. 
Could you disable all modules and check if empty space disappear?

jaykovski, 2020-10-10:
> Hmm, strange behavior.
> Could you disable all modules and check if empty space disappear?

Still empty space with all modules disabled

exelban, 2020-10-10:
Thanks for your help.

I have no idea where is the problem, and why empty space is showing. I was thinking that the problem was in the battery module (macOS report wrong battery status, or I read wrong). But in this case, when you click on empty space, battery popup must shows.

exelban, 2020-10-10:
But I suppose that the problem in the battery module. It can be related to the wrong module initialization. I will recheck all.

exelban, 2020-10-10:
Maybe you will have a few minutes to check a test build?

jaykovski, 2020-10-10:
Yeah sure

exelban, 2020-10-10:
Check please if empty space still shows.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5358887/Stats.dmg.zip)

jaykovski, 2020-10-10:
It disappeared, the strange thing is, I have put the release app back and there is still no empty space, I even restarted the mac just in case and no space. Did it add some preference flag or something?

exelban, 2020-10-10:
Yeap, it checks if the module is unavailable. If yes, it makes hard disable. I suppose this bug appears when I testing a different way to disable/hide widget in menu bar.

Thanks for your help. I will deploy a new version with this fix tomorrow.

exelban, 2020-10-11:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.7) release.

closed 2020-10-11 by exelban