exelban/stats #99: Request: Combined view, update interval, donate button

iliakukharev, 2020-10-05:
- Device: [e.g. Macbook Pro 2019]
 - MacOS [e.g. 10.15.7]
 - Application version [e.g. 2.3.6]

Perfect and fast alternative to iStats, finally.

What I'm missing:
1. Combined view for all metrics, to drag them as 1 item
2. Update interval and switcher to Mb/s for Network, like 0.5Mb/s, updating every 5 seconds
3. Donate button or premium features — we need to have an option to support developer.

exelban, 2020-10-05:
Hi.

There will be nice to have all these points as separate issues.

1. Combined view for all metrics will be necessary in big sur (because there have soo big spaces between menu bar items). So it will be implemented, but only when Big Sur will be available (something can change in API, etc).
2. Update interval for the Network module is a little bit strange for me. Because there is a sense to show real value, not average in the last 5 sec. But it can be done.
3. Yeap, switcher to the KB/MB/GB/TB can be added.
4. I never think about donate button. I don't even know which service I can use for this (for me github sponsorship is ok, but you need to have a gh account for this). Maybe Patreon? It allows you to make cycle payments not one time. Maybe someone know what I can use for this?

nyanev, 2020-10-16:
A combined view is a must-have.

exelban, 2020-10-19:
Hi.

Points 1 and 4 will be added. 2 and 3 will not be implemented. 2 you can see [there](https://github.com/exelban/stats/issues/113). 3 no make sense, app makes it automatically depends on current bandwidth. And adjusting these to the size.

I split these issues into tasks and add them to the pipeline.

closed 2020-10-19 by exelban