exelban/stats #101: CPU Usage

brokenthumbs, 2020-10-06:
**Describe the bug**
Why does the Stats app use high amounts of CPU% ? 
Is this normal ?

[image]

[image]

**Details:**
 - Device: MacBook Pro (13-inch, 2017, Four Thunderbolt 3 Ports)
 - MacOS Version 10.15.1 (19B88) Catalina
 - Processor 3.1 GHz Dual-Core Intel Core i5
 - Memory 8 GB 2133 MHz LPDDR3
 - Graphics Intel Iris Plus Graphics 650 1536 MB
 - Stats Application version 2.3.6

exelban, 2020-10-06:
Hi.

Nope, it's not normal. Does it happened after app restart?

brokenthumbs, 2020-10-06:
Restarting the app seems to have fixed it.

What would cause the CPU Usage for Stats to spike like that, and is there a way to prevent it from happening again?

brokenthumbs, 2020-10-06:
Seems like there might be some sort of leak?

[image]

exelban, 2020-10-07:
Yeap, it can be some leak in readers. Could you help to detect it? Try to disable modules and see when leak will appear/disappear. Unfortunately, I cannot reproduce this leak. I suppose its related to hardware.

brokenthumbs, 2020-10-07:
Sorry, deleted the application. Couldn't have it spiking my CPU randomly. I can't imagine I'm a unique case either, with fairly standard hardware. For what it's worth, the CPU seems to spike after running for about what seems to be an hour. I don't have any tangible proof to offer. Good luck! Hope you can find the bug.

exelban, 2020-10-07:
ok, thanks for reporting.

Maybe someone who will have the same bug could help. If someone will have the same bug, please reopen this issue.

closed 2020-10-07 by exelban

ranliu98, 2021-01-14:
Same issue here

exelban, 2021-01-15:
> Same issue here

Hi. Could you please provide a little bit more info?

mrdylanyin, 2023-04-19:
Same issue, after reboot Mac get fixed.