exelban/stats #102: Request: Add CPU Frequency

matis-schotte, 2020-10-07:
Great little app, good work 👍 
I think I am only missing the CPU freq under the sensors ... that would be a nice little addition.

Also I noticed that usually I see different values in the menu bar vs the drop down (eg cpu usage etc)

exelban, 2020-10-07:
Hi. 
I really want to have CPU frequency. But there is one big problem. You must have installed Intel Power Gadget. And not all have it (only devs I suppose).

There are different sensors of the CPU temperature. It depends on which sensor Intel Power Gadget is using to show.
Nope, I'm not using Intel Energy API, because it must be installed on the system.

exelban, 2020-10-07:
I was tried to add the optional CPU frequency (as in iStat menus). But without success.

matis-schotte, 2020-10-07:
I see, thanks for your reply!

First Idea I had was creating a small helper tool which could be linked to the intel framework. If the framework was installed, the helper tool could report the cpu freq.

After  googling there seem to be other, better approaches around, eg adding the framework to Stats and setting the linking of the framework to optional; or creating a wrapper framework which links to the real one.

https://stackoverflow.com/questions/38536956/conditionally-import-a-framework-such-as-speech-based-on-ios-version-in-swift/38538861

https://github.com/nicholascross/iOSDynamicLoaderExample

exelban, 2020-10-07:
Thanks for the example, I will take a look.
I cannot embed intel framework, it must be installed on the system.

matis-schotte, 2020-10-07:
Yes, sure - but Stats could load and use it if it is already installed on the System, like iStats is doing it.
Maybe I can help when I have free time :) Lets see

exelban, 2020-10-07:
Yeap, but normally you don't have it installed. But if I will make it works optionally (when IPG installed), it will be great.
I will try to take a look at this. Thanks.

matis-schotte, 2020-10-08:
I did try it locally, works with weak linking and then in the app you can check if the eg PG_Initialize() function is available.

https://stackoverflow.com/a/17205070/3078330

exelban, 2020-10-14:
@matis-schotte hi. Do you maybe know how to make a `weak_framework` linking?

When I specify `-weak_framework IntelPowerGadget`, and it not present in the system. XCode throws: `ld: framework not found IntelPowerGadget `.

matis-schotte, 2020-10-14:
Sure! Did you understand my last comment from 6 days ago?

If you call any of the IPG framework methods, it _will_ try to load the framework. So you need to make sure to check for availability first. I saw another app where they were checking for the header file in the framework directly (file exists /Library/Extenstions/Intelblalba...), but I would suggest checking the function pointer != NULL

matis-schotte, 2020-10-14:
Also you can set the "optional" attribute for the linking in xcode, but it shouldnt matter how you do that weak linking

exelban, 2020-10-14:
Yeap, how to make it optional in code I figured out. But now I cannot build the project with and without the framework.

exelban, 2020-10-14:
[Here](https://developer.apple.com/library/archive/documentation/MacOSX/Conceptual/BPFrameworks/Concepts/WeakLinking.html) Apple says that will be better to remove it from the list of frameworks. And add it as `-weak_framework`.

But I when I add it, I throw this error(

matis-schotte, 2020-10-14:
Yeah sure, you _always_ will need the framework to _build_ the project, of course.
But the app will start without the framework. You can simulate that by renaming the framework, and then starting the app.

matis-schotte, 2020-10-14:
If you add the weak linking you cannot build even if you have the framework? If you have commited your work I could probably take a look.

I could get it working by adding the framework in xcode and setting it to link optional. I did not use the `-weak_framework IntelPowerGadget`, but then I guess you may need to update your framework header search paths probably?
Hopefully I understood you now correctly ;)

exelban, 2020-10-14:
Ok, I build the project with presented IPG framework. Run the app, and it works. After that, I remove the IPG (rename). And when I try to run the app it crash. Crash logs says:
```
Dyld Error Message:
  can't resolve symbol _PG_Initialize because dependent dylib #1 could not be loaded
```

matis-schotte, 2020-10-14:
Well it says "dependent" so its not optional/weak linked

exelban, 2020-10-14:
Yeap, but I link it with `-weak_framework`. Ok, I will recheck with optional linking. But as I saw, it just uses `weak_framework` flag when it's optional.

matis-schotte, 2020-10-14:
Maybe this helps?
https://stackoverflow.com/questions/57168931/optional-linking-for-swift-combine-framework-in-xcode-11?rq=1

exelban, 2020-10-14:
the same crash error(

exelban, 2020-10-14:
I already try all these methods(

matis-schotte, 2020-10-14:
Okay, as I mentioned above - commit to a new branch here on github, and give me a link, I can take a look

exelban, 2020-10-14:
Maybe you can help me to understand how it works.

When we specify the framework as `optional`. Run the app, it's running it and tries to find this framework (header file?). 
And the app must be started even without this framework in the system?

exelban, 2020-10-14:
Ok, give me a few minutes.

exelban, 2020-10-14:
Thanks for your help. There is a branch: [cpu_freq](https://github.com/exelban/stats/tree/cpu_freq).

matis-schotte, 2020-10-14:
Okay I tried to compile but I cant seem to find the `Repeater` module (its not added in the xcode project `Swift Packages`)

Anyways looking at the code you are calling PG_Start() without checking if the framework actually is available/if the function pointer is not null ... Or do you have that check somewhere else?
The header file check `#if __has_include(<IntelPowerGadget/PowerGadgetLib.h>)` looks like a compile-time check, not run-time check. This will always be true.

Could you please add the module `Repeater` to the project? Or where is it?

[image]

[image]

matis-schotte, 2020-10-14:
Also - if you like - you can write all that code in swift! The IPG framework can be accessed without problems via bridging header ... works fine

exelban, 2020-10-14:
You can run `make dep`. But you need to have installed [carthage](https://github.com/Carthage/Carthage).
Yeap, I will rewrite it to swift. I just have already this reader written in C in another project.

matis-schotte, 2020-10-15:
- TL;DR:
its still the same, you are calling `PG_start()` in CPU > main.swift:init:56 without checking if the framework is available, as soon as you call _any_ method from the framework it will be loaded which crashes the app if the framework is not available

Solution: Do a _runtime_ check for the framework (actually repeating myself here) this could be something like:
`NSFileManager fileExists at path:...`
or even better check for the null pointer:
```
#import <IntelPowerGadget/PowerGadgetLib.h>
extern bool PG_Initialize() __attribute__((weak_import));
static inline bool intelPowerGadgetIsAvailable() {
	return (PG_Initialize != NULL);
}

intelPowerGadgetIsAvailable() // -> use this before calling any of the framework methods
```
Your _compile time_ check `#if __has_include(<IntelPowerGadget/PowerGadgetLib.h>)` is useless in this regard

**Again, do not attempt to call any IGP framework functions if you don't know if it is available at runtime**

- had big problems disabling signing to build on my machine, gave up finally

- noticed that the framework folder is 100 MB in size in the final product; thats because you seem to include all frameworks in each other (eg Battery > ModuleKit & StatsKit; Inside that ModuleKit there is another StatsKit and Repeat; same for probably each of the others like CPU.framework) -> this currrently seems to be messed up, you ideally only would need each framework once linked to the app

- both carthage frameworks do support Swift Package Manager and could be added directly in Xcode to smoothen initial setup

Good luck 👍

exelban, 2020-10-15:
Ohh. I was doing it wrong.
I finally make it works with and without a framework in the system. As you say, `__has_include` is a compile time check. And it's a reason why it does not work. This snapshot just works perfectly.

Yeap, I will migrate to the swift package manager, it's just some old part stays. And will take a look at how to make a proper linking of frameworks inside the app.

Thanks again for your help. CPU frequency will be available in the next release.

matis-schotte, 2020-10-15:
Very good 👍 And thank you too for your effort in opensource!

phu54321, 2020-10-15:
If one doesn't have CPU frequency, we could use `RDTSC` as an approximation for CPU frequencies. See https://stackoverflow.com/questions/9948987/detect-current-cpu-clock-speed-programmatically-on-os-x for more info

exelban, 2020-10-15:
I checked it. And it shows strange numbers. 99% of the time it shows 3.8GHz. At the same time, the intel framework reports 4+ GHz. My CPU has a 3.8GHz max frequency. So it looks like it shows the frequency without a turbo boost. So it's not a reliable way to measure a CPU frequency. As I know there is only one way to measure it: Intel Power Gadget.

exelban, 2020-10-18:
Added in the [last](https://github.com/exelban/stats/releases/tag/v2.3.8) release.

closed 2020-10-18 by exelban

matis-schotte, 2020-10-19:
Great work! I did update to the latest but dont see any cpu frequencies ... where are they located?
I did look in the sensors module but still only have Temp, Volts and Watts ... ?

exelban, 2020-10-19:
In the popup view. Near the load.
[image]

medack, 2023-10-14:
Sorry for resurrecting this old issue, but Intel recently discontinued the Power Gadget (https://www.intel.com/content/www/us/en/developer/articles/tool/power-gadget.html), recommends uninstalling it and points now instead to the PCM API (https://github.com/intel/pcm). Could that now be used by stats?

richardzone, 2023-11-08:
@medack correct me if I'm wrong but PCM is difficult to install on MacOS, as no precompiled binary is published and Apple Developer ID is needed to properly sign the kext (or disable SIP which is not ideal)

medack, 2023-11-08:
I honestly was not aware of that, but they seem to provide only the source code, so you are totally right.

The only case scenario which I can think of where that would not be a problem is for the people that use OpenCore to boot unsupported macOS versions on old Intel machines. Those will already have their SIP lowered, which would allow the kext to boot. Otherwise, I assume it would be very difficult to implement it, yes. Nevertheless, should the Power Gadget recommendation stand as it is?

Datisit, 2023-11-14:
Just want to chime in here to say I'm in that exact boat.
I'm not an active user of Stats yet (I currently still run iStat Menus), however as iStat seems to be end-of-life I would consider jumping to Stats once it's ready. 

I'm on a Hackintosh, and have recently uninstalled the Intel Power Gadget partly due to troubleshooting some system instablilty and partly as I noticed Intel had stopped supporting it.

As they recomended PCM, I have just downloaded, compiled and installed it. And my next thought was, well I doubt iStat Menus is going to incorporate this to get the frequency meetering back into the CPU module.

So if Stats did that, that would be another great reason to jump over for anyone else in my position.

Thanks.