exelban/stats #103: White background on adding/changing stat icons

osavchenko, 2020-10-07:
**Describe the bug**

When I add a new stat icon (like sensors, CPU) or changing it's layout white background can apperas. And when I continue switching icon state (enable/disable) or changing its layout background can get back to normal.

Temperature information on white background: 
[image]

**Details:**
 - Device: Macbook Pro 2012
 - MacOS 10.15.7
 - Application version 2.3.6

exelban, 2020-10-08:
Hi. 

Thanks for reporting. I will take a look at this.

exelban, 2020-10-11:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.7) release.

closed 2020-10-11 by exelban