exelban/stats #104: All CPU cores only shown when show hyperthreading cores enabled on a CPU without hyperthreading

igorkulman, 2020-10-07:
**Describe the bug**
I run the app on a hackintosh with [i5-3570K](https://ark.intel.com/content/www/us/en/ark/products/65520/intel-core-i5-3570k-processor-6m-cache-up-to-3-80-ghz.html). This CPU has 4 physical core and no hyper-threading. 

I would expect the app to show 4 cores when "Show hyper-threading cores" is disabled, because there are 4 physical core. It only shows 2 cores. When I enable "Show hyper-threading cores" it shows 4 core but it does not make much sense.

**Details:**
 - MacOS 10.15.7
 - Application version 2.3.6

exelban, 2020-10-07:
Hi.

Thanks for reporting. I will fix this.

igorkulman, 2020-10-07:
If you need anything for me, for example logs or output of something like `sysctl -a | grep cpu` just let me know

exelban, 2020-10-11:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.7) release.

closed 2020-10-11 by exelban