exelban/stats #105: Thick borders on low resolution displays

igorkulman, 2020-10-08:
**Describe the bug**
When I run on the app on a low resolution display in my case, the borders around the sensors are too thick compared to when running on a high resolution display.

1600x900

[image]

4K

Would be nice to have the border say 1pt only on a low resolution display to make it look the same thickness as on a high resolution display.

**Details:**
 - Device: Thinkpad T440s (hackintosh)
 - MacOS 10.15.7
 - Application version 2.3.6

exelban, 2020-10-10:
Hi.

I'm not sure how it can be fixed. Low resolution display can be caught by PPI. 
I already using 1px for these borders. I'm not sure if it will be better if I set <1px. Maybe there is some smoothing which I can enable for low resolution display. I must make some research on this.

igorkulman, 2020-10-10:
It should be possible, other apps can draw a line that looks OK. The battery icon on this screenshot is from [Battery Monitor](https://apps.apple.com/us/app/battery-monitor-health-info/id836505650?mt=12)

[image]

But I get that it is not easy and only effects a few people with old Macbook Airs.

I wanted to try different values for width but I cannot get Carthage to build the dependencies.

tech189, 2020-11-03:
I would also appreciate this fix because I use my laptop plugged into a 1080p display. This also affects the battery icon, see the screenshot, which shows the built-in battery icon for comparison:
[image]

closed 2020-12-06 by exelban

luckman212, 2020-12-06:
Thank you 🙏🏻 so much for this fix! I must be in that 0.01% group that also still use a low-res 1080p display (Mac Mini) so this is a very appreciated change.